mavlink/qgroundcontrol #8208: Strange altitude sent in follow mode

twistedwings, 2020-01-08:
I switched to follow mode and walked along a flat area.
This is the altitude ArduCopter got:

[image]

see also [https://discuss.ardupilot.org/t/strange-altitude-in-follow-mode/50781/4](url)

ArduCopter 4.0.0
QGC Android daily 2020-01-03

DonLakeFlyer, 2020-01-09:
I'm going to need to add some additional logging to the builds to figure out what is going on here. Hope to get that some in a day or so.

Thanks for the bug reports.

Antiheavy, 2020-01-11:
I assume QGC is using the location services from the device (some android device in this case)?  Depending on the quality of your device's GPS signal, it could be possible that your device's altitude estimate is really that terrible...  or maybe its a bug...

DonLakeFlyer, 2020-01-15:
> I assume QGC is using the location services from the device (some android device in this case)? Depending on the quality of your device's GPS signal, it could be possible that your device's altitude estimate is really that terrible... or maybe its a bug...

That is correct as is the comment on crap GPS Alt. I hope to have time soon to add logging of location info in QGC so that it can be verified to be that or some other problem.

DonLakeFlyer, 2020-01-16:
@twistedwings New daily builds should be available in an hour. Can you use the new version. Turn on FollowMeLog console logging. Capture the output while moving around and attach here? This should tell me what is really being reported as altitude from the device.

twistedwings, 2020-01-16:
may take a while as it's raining all day long.

twistedwings, 2020-01-17:
Where do I enable FollowMeLog? I cannot find that option.

[image]

dogmaphobic, 2020-01-17:
Under "Console" in _General Settings_:

[image]

twistedwings, 2020-01-17:
I don't have that, do I need a newer version?

[image]

dogmaphobic, 2020-01-17:
It's in the current master (daily) build.

twistedwings, 2020-01-17:
I've got the 64 bit APK from hier: [https://docs.qgroundcontrol.com/en/releases/daily_builds.html(url)
Option is not there.
Which version do you have?

dogmaphobic, 2020-01-17:
I just built it myself. I'm not sure how frequently daily Android builds are run.

twistedwings, 2020-01-17:
It's in the windows version but not in the android version.

dogmaphobic, 2020-01-17:
You said "APK", which implies Android. That's what got me confused. Either way, the change went in two days ago. It should be in the current daily. If it isn't, we have to wait for @DonLakeFlyer.

DonLakeFlyer, 2020-01-18:
Android daily builds should post with ever pull merged. Looks like something broke with Android on the 15th. Nothing has been posted since then.

DonLakeFlyer, 2020-01-18:
As far as I can tell android builds just spontaneously combusted as opposed to some specified pull screwing them up. Will likely have to wait until I get back for me to figure out why.

dogmaphobic, 2020-01-19:
He's using Windows. I only got confused by this:
>I've got the 64 bit APK from hier: 

As to the Android build, it looks like something changed with gradle. If you don't already have it, the build downloads it for you. We all have it so we are probably running an earlier version. As the CI build is always from scratch, it is downloading something new and that gradle seems to be messed up.

twistedwings, 2020-01-19:
No, I am using Android. The whole issue is about Android.

dogmaphobic, 2020-01-19:
>No, I am using Android. The whole issue is about Android.

>It's in the windows version but not in the android version.

That can be confusing :P

Yes, the daily Android build hasn't run in some time. We need to wait for @DonLakeFlyer.

twistedwings, 2020-01-25:
Now the option appears in the daily.
I made a test with version 2020-01-24.
I don't see any FOLLOW logging in the telemetry log.

[2020-01-25 10-41-59.tlog.txt](https://github.com/mavlink/qgroundcontrol/files/4111941/2020-01-25.10-41-59.tlog.txt)

DonLakeFlyer, 2020-02-08:
It's not a telemetry log, it's a console log which you can save from Settings/Console

DonLakeFlyer, 2020-02-11:
@twistedwings Possible to get a console log for this? I'm trying to wrap up all the problems for a Stable release.

twistedwings, 2020-02-11:
It may take a while until the rain and storm stops.

twistedwings, 2020-02-15:
here it is, sorry for the delay.

[Follow2.txt](https://github.com/mavlink/qgroundcontrol/files/4208659/Follow2.txt)

DonLakeFlyer, 2020-02-15:
So what I think this is showing is that the altitude value from a tablet gps can be total crap. Which in turn means using relateive mode heights will pretty much never work. Where as absolute should work fine. I"m going to change QGC to only support Absolute height.

DonLakeFlyer, 2020-02-15:
FYI: I've tried the same thing with multiple devices and the altitudes are all over the map. With an iPhone being the closest. And for me most Android tablets showing negative altitudes for my test area which is way your tablet is doing as well.

twistedwings, 2020-02-15:
Yes, I totally agree with that. Although I think disabling relative mode is not good as some users may want that. Is it possible to support baro if installed?

DonLakeFlyer, 2020-02-21:
Working correctly now based on new home altitude only support

closed 2020-02-21 by DonLakeFlyer