mavlink/qgroundcontrol #8212: On the radio page, the sliders that show joystick movements do not move along the full range

gittymcgitface, 2020-01-12:
## Expected Behavior
After calibrating my throttle, when I move it fully up and fully down, the circle on the slider that relates to this channel on the radio page should likewise move fully to the left of the slider and fully to the right.

## Current Behavior
On the radio page, full range movements of the controls on the transmitter do not result in full range movements of the circle graphics on the sliders.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. With a transmitter connected to a Pixhawk 1 via an X8R, click on the "Radio" tab.
2. Move the transmitter controls fully up and down.
3. Observe that the GUI does not show the circles to move across the full range of the slider.

## System Information
When posting bug reports, include the following information
- Debian Linux
- QGC Version: 3.5.6
- QGC build: AppImage
- Flight Controller: Pixhawk I
- Autopilot (with version):  Ardupilot Copter 4.0.0

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

https://imgur.com/wsuYYgE

DonLakeFlyer, 2020-01-15:
Although what you describe would make some amount of sense. It isn't the way it works. The indicators just show a raw PWM value in the range of 800<->2200. Calibrating **has no affect** on that. Calibration **does have an affect** on whether the channel is reversed or not. Once direction is detected the direction should be correct.

closed 2020-01-15 by DonLakeFlyer

gittymcgitface, 2020-01-15:
Okay, thanks for responding.

I'd suggest labelling the axes in the sliders to avoid confusion in the future.