mavlink/qgroundcontrol #8213: Pixhawk 2.1 Cube - Error: Sync: Send Command: Invalid sync response: 0xfe 0x3c

TobiMiller, 2020-01-13:
Hey,

i can flash the pixhawk cube with the px4 stack in mission planner. but with qgroundcontrol it is detected als FMU V2 instead of FMU V3 and it says:

Error: Sync: Send Command: Invalid sync response: 0xfe 0x3c

is this a known bug?

TobiMiller, 2020-01-15:
daily build worked in the end. i dont know what but the procedure to get a brand new cube running with PX4 Stack has no out of the box experience.
with ArduPilot and MissionPlanner everything is working. Flashing PX4 Stack with MP lets the FC beeping all the time on boot.

closed 2020-01-15 by TobiMiller

dlwalter, 2020-02-25:
I am running into this same issue.  Why was this closed?

DonLakeFlyer, 2020-02-25:
> daily build worked in the end.

This?

DonLakeFlyer, 2020-02-25:
You can reproduce with latest daily?

dlwalter, 2020-02-25:
Hey Don,

I'm seeing this with 3.5.4, 3.5.5, and 3.5.6 on a PX4 Cube Black on macOS 10.15.2 using the v1.9.2 firmware.  No problem uploading from the command line.

Haven't had time to try with the daily.

DonLakeFlyer, 2020-02-25:
Current Stable 3.5.* builds don't fully support the latest chibios and/or ArduPilot bootloader stuff. Tons of updates/fixes for that have gone in since last Stable.

First phase of new Stable 4.0 is rolling out as I type. Unannounced at this point but if you download from the web site you'll get new 4.0. Try that and let me know what happens. I'm letting 4.0 bake for a day to pick up any last minutes fixes. And then I'll mark so QGC it will announce there is a new build and post release notes and so forth.

Note: OSX 4.0 is not yet available due to some build CI problems.