mavlink/qgroundcontrol #8216: CameraTriggerIndicator not working

moreba1, 2020-01-13:
Hi 

I work with px4 code for survey application. i can not see camera trigger indicator in qgc flight display menu but before when i was tested with ardupilot code it was working.
i checked tlog file of px4 code , The location of the camera triggers was saved , so there is a problem .

Antiheavy, 2020-01-13:
I think mavlink 2 is required for PX4. Are you enabling mavlink 2 on your system?

DonLakeFlyer, 2020-01-15:
> i checked tlog file of px4 code

@moreba1 So in your telemetry log file (not a vehicle binary log) you played it back in QGC and you can see the MAVLINK_MSG_ID_CAMERA_IMAGE_CAPTURED messages in the mavlink inspector sent from the vehicle to QGC? But the indicators still don't show up in the Fly View?

moreba1, 2020-01-16:
i checked ma

moreba1, 2020-01-16:
@DonLakeFlyer 
i checked mavlink inspector .

[image]

DonLakeFlyer, 2020-01-18:
QGC will only show image capture points if CAMERA_IMAGE_CAPTURED.capture_result == 1 which indicates a successful capture according to mavlink spec. In the screen shown above you can see that CAMERA_IMAGE_CAPTURED.capture_result equals -1 which isn't even a valid value for that field. So as far as I can tell QGC is working correctly.

https://mavlink.io/en/messages/common.html#CAMERA_IMAGE_CAPTURED

moreba1, 2020-01-25:
Does it possible to display camera indicator with different color when result is 0?

closed 2020-11-25 by moreba1