mavlink/qgroundcontrol #8220: Survey item terrain following: Max climb/descent rate not applied to turnaround waypoints

sanderux, 2020-01-15:
See https://validator.deltaquad.com/validator/3d.php?uid=5e1ecbf7bbfdb5e1ecbf7bc01d&flightdate=1579071600&localFlightdate=1579075200

When planning a survey on a hill with terrain following you can set a max climb/descent rate. This is only applied to the transects within the mapping mission. The turn-around waypoints do not consider this limitation as shown below:

[image]

moreba1, 2020-01-16:
fixedwings in turnaround must do a loiter to alt (next line alt) if needed .

sanderux, 2020-01-16:
> fixedwings in turnaround must do a loiter to alt (next line alt) if needed .

That would completely screw up a clean entry into the next transect. Also, it defies the purpose of setting a max climb/descent rate as you might hit a mountain while circling up.

sanderux, 2020-01-16:
@DonLakeFlyer actually setting a max/min gradient would be more use friendly

DonLakeFlyer, 2020-02-08:
Ok, so this is a pretty hairy problem to solve. It won't make it into stable which is being rolled out now. But I'll pick it up right after that.

moreba1, 2020-05-28:
@sanderux 
this type of mission can solve the issue and it is better for photogrammetry.

[image]

DonLakeFlyer, 2020-12-10:
@sanderux I'm working on this new. Do you happen you have a good Plan file which I could use for testing which shows the usefulness of this?

sanderux, 2020-12-13:
In the validator link on top there is a menu with an option to download the plan file.
does that help?

sanderux, 2020-12-13:
You could even use the DQ validator to check if your modifications solve it

sanderux, 2020-12-13:
@moreba1 we do not allow such behavior as the path is unvalidated and unpredictable

moreba1, 2020-12-13:
> unpredictable

the plane can change more altitude during turning between lines, we can define max altitude change in turn.
when the distance between lines is low, setting Max climb/descent for turning is not appropriate because the turning trajectory for the plane is longer than waypoint lines and the plane can change more altitude.

sanderux, 2020-12-13:
that all works nice if you are above flat land, but when given a max climb/descent it should be honored imo, you could crash into a mountain when loitering up/

closed 2020-12-13 by DonLakeFlyer

DonLakeFlyer, 2020-12-13:
> In the validator link on top there is a menu with an option to download the plan file.
> does that help?

@sanderux I was wondering if you already had some problematic plans for which this change would now make them work.

sanderux, 2020-12-13:
@DonLakeFlyer yes, the plan in de main issue description is problematic. Thats why i refered to it. I'm traveling now so a little hard to upload it here but it is in the link on top

DonLakeFlyer, 2020-12-13:
@sanderux The validator link for that is no longer valid. So I can't figure out where that is so I can replicate it.

sanderux, 2020-12-13:
Ah, i see now, let me try and make you one in a few hours

PR #9249 (2020-12-13): Plan: Support terrain adjustment in turn segments
* Previously only the survey transect was terrain adjusted. The turn was not.
* Along the way of reworking this I fixed a bunch of terrain profile display bugs. There are still a bunch there though which I will com back to.
* Fixed a pile of toolstrip button problems
* Returned terrain elevations not average (sort of) across the four nearest know values of elevation data. This leads to smoother terrain data.
* Fix #8220

Here you can see it in action in a simple two transect survey:
[image]