mavlink/qgroundcontrol #8221: Launch position moves when VTOL Takeoff item is moved

sanderux, 2020-01-15:
Not all mission plans consider the VTOL_TAKEOFF identical to the launch position.
After setting the launch position manually it should no longer move

DonLakeFlyer, 2020-01-15:
That is how it is supposed to work. So bug.

DonLakeFlyer, 2020-02-08:
> Not all mission plans consider the VTOL_TAKEOFF identical to the launch position.
> After setting the launch position manually it should no longer move

No wait. That is the way it is supposed to work. Hmm...

DonLakeFlyer, 2020-02-08:
Does PX4 VTOL takeoff work like MultiRotor. For multi-rotor even if the takeoff location differs from home position it still goes straight up above home to altitude and then finally flies to the takeoff location.

DonLakeFlyer, 2020-02-08:
> After setting the launch position manually it should no longer move

How specifically are you doing that? Want to understand sequence of steps.

pompecukor, 2020-02-16:
> > Not all mission plans consider the VTOL_TAKEOFF identical to the launch position.
> > After setting the launch position manually it should no longer move
> 
> No wait. That is the way it is supposed to work. Hmm...

Hi Don,

A bit related. This has been my main quibble with QGC. Take off and Vtol take off, are not supposed to have coordinates. Or better yet coordinates are irrelevant. Home position is another thing.
I see this issue all the time. Especially when mission is planned with mission planner.
If later we connect to QGC and download. it would show Home at the correct position, but show the take off a cord 0.0. Somewhere in the ocean.
Take happens wherever you put the plane. Why do you make qgc insiste on having a valid coordinate for it, when in fact it doesn't use one?
Could you not fix this somehow?
Mission planner (visuall)  just sets take off to where home is set. From there you get a dotted line to the first way point. After that everything is solid.
Could you not just set 0.0 to fallbank to home? I mean that is what will happen anyway. Just QGC giving the wrong visual representation.

Thank you

sanderux, 2020-02-16:
> > > Not all mission plans consider the VTOL_TAKEOFF identical to the launch position.
> > > After setting the launch position manually it should no longer move
> > 
> > 
> > No wait. That is the way it is supposed to work. Hmm...
> 
> Hi Don,
> 
> A bit related. This has been my main quibble with QGC. Take off and Vtol take off, are not supposed to have coordinates. Or better yet coordinates are irrelevant. Home position is another thing.
> I see this issue all the time. Especially when mission is planned with mission planner and upload to plane.
> If later we connect to QGC and is syncs the mission (downloads it), it would show Home at the correct position, but show the take off a cord 0.0. Somewhere in the ocean.
> Take happens wherever you put the plane. Why do you make qgc insiste on having a valid coordinate for it, when in fact it doesn't use one?
> Could you not fix this somehow?
> Mission planner (visuall) just sets take off to where home is set. From there you get a dotted line to the first way point. After that everything is solid.
> Could you not just set 0.0 to fallbank to home? I mean that is what will happen anyway. Just QGC giving the wrong visual representation.
> 
> Thank you

Yes they do according to spec. Both the takeoff and the vtol takeoff can be operated as only the takeoff command, or as the first waypoint. The reason for this is that missions are then able to be planned in advance when the exact home position is not yet known.

The software can deal with both situations and will consider the takeoff complete directly above home if it is inside the acceptance radius.

pompecukor, 2020-02-16:
I don't know if what you are saying is PX4 specific.
As far as Arduplane (ans mission planner) goes There always a home.
As far as the vehicle itself goes take off and Vtol take will always coordinate-less. Simply reason being that take off launch will start from wherever you are when you actually lunch the plane and (b) Vtol take off is always 100% vertical.
Also no matter where you actually put your home in the mission, it will always update to the actual position of the platform when you are it.

Bottom line is, there is no position for it on the vehicle, so why force a position for it on the mission planner?

All this does is make a messed of the mission when moving between MP and QGC. And we both know which is more popular. So what is the big dél (or loss felt) if it was to asher to the version with the bigger practice user base.

Setting a home point in your mission, before you go on the field make a lot of sense. We do that al the time. Put it around the general location that the flight is planned. 

However forcing a coordinate for the take off makes no sense. So why do it?

sanderux, 2020-02-16:
Considering i implemented both vtol_takeoff and vtol_land i'm pretty sure what the intention was. I remember back when i was coordinating the vtol stuff with tridge to be a little consistent between px4 and apm we spoke about this. If APM doesnt navigate to the waypoint on coordinate basis then APM is not following the spec.

If you need a simple takeoff and transition those commands are available. vtol_takeoff is certainly a coordinate based waypoint (or it should be).

pompecukor, 2020-02-16:
> Considering i implemented both vtol_takeoff and vtol_land i'm pretty sure what the intention was. I remember back when i was coordinating the vtol stuff with tridge to be a little consistent between [...]

Well you will be surprised that is it not.

The way Vtol_takeoff happens on arduplane (and Mission Planner mission planning mimics this as-well) is that the unit transition if there is a waypoint for it to fly to.

That is all VTOL_takeoff does it tell it to take off and climb to the set AGL. That is the only information that is added to it. The altitude. If there is not proceeding "command" then it would stop at that altitude and automatically switch to RTL which in this case will make it just land on the same spot as it is withing the "home radius".

It is actually way more intuitive this way, than what you have described.
It the most basic case (and goal is to keep it simple) you never ever need to tell the VTOL to transition. That thing QGC mentions in the takeoff command is just there for looks. It is the Firmware on the unit that does the control. So that is what must be respected.

All you need to tell the plane is VTOL take off to 40m (just as example).
Then next command can be a waypoint with coordinates. You can even use things like Loiter to alt immediately as a waypoint.

The unit will know to transition and start heading for that waypoint. You do not need to instruct it to transition.

All in all, I still do no get why you think the vtol_Takeoff needs to be a coordinate based thing? Even if you needed to add a manual transition in it. Still make zero sense.

You drop the plane where you want. Acceding sideways if typical very dangerous and not desirably. So it will always ascend exactly vertically. So what sense does any coordinate make? None!

As a matter of fact in Arduplane it is even a safety feature that if you do add coordinates to the vtol takeoff. When you write the command the plane with will reset them to 0,0. Just to avoid taking of sideways.

pompecukor, 2020-02-17:
[image]

The above is how it works. As I described above.
Just those two command lines.
First is the the rising to altitude (40m) and the second as it is a remove waypoint will make it transition and fly there. Of course there are paramteres inside of arduplane if you do not want to transition while flying to there, so you can any amount of complex mission. This is just the baseline and what makes most sense.

And this is exactly the same for fixed wing take off. Only difference is that you have a pitch value as well in the takeoff line. Again, the plane does not care where it is. I will take off from where every you put it and in what every direction you point it. Then when it gets to the altitude that you set. Only after that will it care about what comes next. It is very logical.

pompecukor, 2020-02-17:
> 
> 
> Does PX4 VTOL takeoff work like MultiRotor. For multi-rotor even if the takeoff location differs from home position it still goes straight up above home to altitude and then finally flies to the takeoff location.

you can see it is same for copter to as Don has mentioned here...

DonLakeFlyer, 2020-02-17:
> A bit related. This has been my main quibble with QGC. Take off and Vtol take off, are not supposed to have coordinates. Or better yet coordinates are irrelevant. Home position is another thing.
> I [...]

I can certainly fix it. And I did a few months ago for both fixed wing and VTOL when someone reported it. I can't fix things people don't tell me about.

Also to clarify:
* PX4: TAKEOFF and VTOL_TAKEOFF takeoff specify a coorinate. As far as I understand multi-rotor and VTOL vehicles will go straight up from home to takeoff altitude and then go to specified takeoff location. Or at least I think this is what VTOL does, but not sure which is why I asked the questions.
* ArduPilot: Does not support a position in TAKEOFF AND VTOL_TAKEOFF. Non fixed wing just goes straight up above home to takeoff alt.

Whether one is correct behavior and one is incorrect behavior doesn't matter to QGC. That's a discussion to have with firmware people. QGC has to deal with reality as opposed to spec.

@sanderux I'll get back to you here soon with what I'm trying to accomplish with the new home/takeoff mission handling in QGC. Then we can discuss further.

pompecukor, 2020-02-17:
> 
> I can certainly fix it. And I did a few months ago for both fixed wing and VTOL when someone reported it. I can't fix things people don't tell me about.
> 

You are just Awesome Don. I did report this about 2 years ago, but it seemed to have been washed away or I did not explain myself well. Or I did not follow up. Thank you.
Will give QGC a go again.

Just a quick question... What does "pause" do for a fixed wing or VTOL. I am scared to try it live and there is no sim in QGC to test safely.  Obviously we cannot pause a fixed wing. If we need to wait, then we switch to loiter (circling at a fixed radius). Vtol could hover, be very dangerous and inefficient, so also loiter is what we use.
Thank you for the insight.
Cheers.

@sanderux 
As it turns out, it was what I suspected that the two firmware where do things differently. Live and let live ;)
However I am still curious to know you answer as to why you think the PX4 way (spec or no spec) makes more sense... for you at least? I still don't understand what the practicality is.

DonLakeFlyer, 2020-02-17:
> What does "pause" do for a fixed wing or VTOL

I changes to a flight mode based on firmware type:
* Copter: Brake
* Plane (which would incloude VTOL): Loiter
* Rover: Hold

> I did report this about 2 years ago, but it seemed to have been washed away or I did not explain myself well.

Just keep asking again if something like this happens. I turn around most things really fast. But the noise level of requests which simply make no sense is pretty high. So things do get lost. Best way for folks who know what they are talking about is create a GitHub Issue.

pompecukor, 2020-02-17:
@DonLakeFlyer 
You are the man :) Thank you.

sanderux, 2020-02-18:
@DonLakeFlyer correct, vtol acts the same as regular MC takeoff. It rises from home to the indicated altitude, transitions to FW and then proceeds to the coordinates (unless these are already in the acceptance radius)

@pompecukor the reason for the vtol-takeoff behaving this way is for consistency with the regular takeoff and also with the vtol-land behavior. I do admit that from a useability perspective there might be some confusion.

pompecukor, 2020-02-18:
@sanderux 
I still don't see the point. Nor have you provided a compelling argument. 
I mean think about it.. How many time out of 100 would you (or any random user) not want the takeoff point to be the home point. Even if you planned the mission at home (which was your first example) it is then expecially true that when you get on the field, you will not place the vehicle on the exact spot where you planned it on a Bing map. There would be a Bush there or something. You see visually where would be a good spot and drop it. So the whole position setting up for take off makes no sense. 

As I said ArduPilot, as soon as vehicle arms makes sure this is the case, by always reseting home to the arming point. This prevents mistakes from unwary users as well as keeps everything "practical". This makes sense for an everyday normal use point of view.

So if the above is the case then "flying to home point" right after take off is never the case. Making any such feature are useless. After you take off, you usually 99 out of a 100, want to go somewhere. And that somewhere is not "home" where you are actually at. But rather your goal, which is set by a "waypoint".

In case of Vtol ArduPilot is smart enough to know to transition for you. So you (user) does not even need to worry about that and can concentrate on more important matters.

sanderux, 2020-02-18:
> I still don't see the point. Nor have you provided a compelling argument.

Consistency *is* a compelling argument

> I mean think about it.. How many time out of 100 would you (or any random user) not want the takeoff point to be the home point. 

I never claimed the takeoff was not always at the home point. It always ascends at the home point. But where it goes from there is the queston

> Even if you planned the mission at home (which was your first example) it is then expecially true that when you get on the field, you will not place the vehicle on the exact spot where you planned [...]

It is exactly for this reason that you can set the takeoff to be the first waypoint, it can be at a large distance from your home point (or planned home point) and will simply be considered the first waypoint. Whereever you decide to launch it from it will always fly towards that waypoint first and then proceed with the mission, This facilitates planning a mission ahead of time (in the office) and being able to fly it without any further modifications. 

> As I said ArduPilot, as soon as vehicle arms makes sure this is the case, by always reseting home to the arming point. This prevents mistakes from unwary users as well as keeps everything "practical". This makes sense for an everyday normal use point of view.

That is also how px4 does it.

> So if the above is the case then "flying to home point" right after take off is never the case. Making any such feature are useless. After you take off, you usually 99 out of a 100, want to go [...]

I think what you are failing to understand is that it will not "fly to the home point", but it will fly to the VTOL_TAKEOFF point, which in many cases is hundreds of meters away from the home point.

> In case of Vtol ArduPilot is smart enough to know to transition for you. So you (user) does not even need to worry about that and can concentrate on more important matters.

And so is PX4, which is why i created the VTOL_TAKEOFF option in the first place.

Just to make clear what i mean, and perhaps for your understanding: this is what a typical mission on our vehicle looks like:

[image]

DonLakeFlyer, 2020-02-18:
> It rises from home to the indicated altitude, transitions to FW and then proceeds to the coordinates (unless these are already in the acceptance radius)

Perfect. Exactly what I needed to know. The fact that it transitions after hitting alttiude is the point I was missing. GIven that, it makes VTOL takeoff way closer to fixed wing takeoff which require seperate launch and finally takeoff location as two separate things. I'll figure out how to fix that.

pompecukor, 2020-02-18:
> 
> 
> > It rises from home to the indicated altitude, transitions to FW and then proceeds to the coordinates (unless these are already in the acceptance radius)
> 
> Perfect. Exactly what I needed to know. The fact that it transitions after hitting altitude is the point I was missing. Given that, it makes VTOL takeoff way closer to fixed wing takeoff which require separate launch and finally takeoff location as two separate things. I'll figure out how to fix that.

Not for Ardupilot it does not. the direction is goes is where ever you throw it to. will fly in that direction till it get to the set altitude, then it will look for next way-point.

You probably know this already, just wanted to make sure.

pompecukor, 2020-02-18:
> I think what you are failing to understand is that it will not "fly to the home point", but it will fly to the VTOL_TAKEOFF point, which in many cases is hundreds of meters away from the home point.

Yes, I my first understanding is that it was the other way around for you guys.
So this is clear now. However I still don't see the "practical" logic.
I have described a typical Ardupilot users use case. What you do with your version does not provide any benefit to us. So I think we can leave it at this and just agree to disagree ;)

DonLakeFlyer, 2020-02-19:
@sanderux Can you verify?

closed 2020-02-20 by DonLakeFlyer