mavlink/qgroundcontrol #8228: QGroundControl taking control of serial port even with AutoConnect off

rijesha, 2020-01-16:
# This system is for posting bugs or feature requests ONLY. Support questions will be automatically closed with no response.
# For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

----
# To post a bug report
When posting bug reports, include the following informaiton:
- Ground station operating system.
- QGroundControl version and build (daily, stable, self-built from source, etc.)
- **If you are using a Stable build which is not the latest, first step is to install latest Stable build and try again prior to entering a bug report.**
- Autopilot board: Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.
- Autopilot firmware (e.g. PX4, ArduPilot, custom) and version.
- Exact steps to reproduce the problem. Starting from booting QGroundControl to when the problem occurs.

**All this template text should be deleted before creating your issue.**
I am using Elementary OS Juno, which I believe is based off of Ubuntu 18.04 LTS. 
I am using QGroundControl v3.5.6.

I have a radio modem connected to the computer via an RS232 to USB adapter (pl2303 converter is what shows up in dmesg)

My architecture involves a python program that interfaces directly to the serial port (usually /dev/ttyUSB0 @57600) and then rebroadcasts out via UDP. QGroundControl connects to the rest of the system via this UDP link, and the python program handles incoming messages from both streams and forwards them appropriately.

What is happening is that the python app will be communicating with the modem just fine, but as soon as I open QGroundControl the link stops and no more messages comes through. If I look in /var/lock I can see a file is made called LCK..ttyUSB0.  As soon as I close QGroundControl that lock file disappears and the python program is able to communicate with the serial port again.

If I try opening the serial port directly with QGroundControl I get this error. 
"Link Error" "Error on link Serial Device on ttyUSB0. Error connecting: Could not create port. Permission error while locking the device"

I have removed modemmanager and my user is part of dialout. This problem seems somewhat recent. The setup seemed to be working properly ~3 months ago. The only hardware change since then is I am using a different modem requiring an RS232 to USB adapter, rather than a TTL to USB adapter. Everything is unchecked in AutoConnect to the following devices.

DonLakeFlyer, 2020-01-18:
First thing is try a daily build instead. Second is turn on LinkManagerLog console logging (https://docs.qgroundcontrol.com/en/SettingsView/console_logging.html). To see why QGC may be trying to connect to the port.

rijesha, 2020-01-20:
I tried the daily build and I think the serial port problem is resolved. My python program is not getting interrupted anymore simply by turning on QGroundControl. I am getting a segfault with the daily build whenever I try to establish a connection.. Ill try again with a newer build as it comes out. 

> QApplication: invalid style override passed, ignoring it.
    Available styles: Windows, Fusion
Settings location "/home/rijesh/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "*Log.debug=false\n"
GStreamer debug output: "/home/rijesh/Documents/QGroundControl/Logs/gst/gstreamer-log.txt"
MAVLinkLogManagerLog: MAVLink logs directory: "/home/rijesh/Documents/QGroundControl/Logs"
System reported locale: QLocale(English, Latin, Canada) "en_CA"
Loading localization for "en_CA"
Qt localization for "en_CA" is not present
Map Cache in: "/home/rijesh/.cache/QGCMapCache300" / "qgcMapCache.db"
Creating default geom
qt.positioning.geoclue2: Unable to start the client: "org.freedesktop.DBus.Error.AccessDenied" "Agent rejected 'QGroundControl' for user '1000'. Please ensure that 'QGroundControl' has installed a valid QGroundControl.desktop file."
QGCPositionManager error 0
Switching outbound to mavlink 2.0 due to incoming mavlink 2.0 packet: 0x1f3a668 1 2
Temp log "/tmp/FlightData146713.mavlink"
Request for unknown firmware plugin factory 5
Request for unknown firmware plugin factory 5
Segmentation fault

closed 2020-02-13 by rijesha