mavlink/qgroundcontrol #8240: CMake transition requirements

dagar, 2020-01-20:
Let's collect all the requirements for a full transition to cmake. Starting with @DonLakeFlyer's points from https://github.com/mavlink/qgroundcontrol/pull/8218#issuecomment-574910104.

* [ ] Build all OS versions in CI, travis and appveyor
* [ ] Can build both Auterion and Sentera custom build as test cases for custom support
* [ ] Can build iOS and android builds which can be pushed to store
* [ ] Android build should be push to store from CI
* [ ] iOS build would still be some manual magic that Gus does. If cmake could automate that as well well even better. But were aren't there yet so that isn't a requirement.
* [ ] Xcode build (Spit out by qmake) plus run QGC in the simulator all the time. So that support can't disappear either.
* [ ] android debug/deploy cycle

dagar, 2020-01-20:
@patrickelectric @tcanabrava

patrickelectric, 2020-01-21:
I can confirm that video is working for both linux and mac

alexeylysenko, 2020-03-04:
Take a look at #8472 it fixes cmake build issue

DonLakeFlyer, 2020-03-04:
Just seeing this now. Good list.

closed 2024-09-22 by HTRamsey