mavlink/qgroundcontrol #8250: Proposal: option to enter jobsite or title for video files

jaxxzer, 2020-01-22:
Use case: a user visits several job sites over the course of the day to collect video. The user would like to be able to easily identify and distinguish which site or job each video corresponds to.

I propose that when the user begins recording a video, a dialog opens where they have the opportunity to enter a custom title for the video file (which can also be included in the telemetry video overlay file), if the user does not wish to enter a custom title, they may hit escape or close the dialog, and the default date+time will be used.

hamishwillee, 2020-01-28:
FWIW I think this popping up would people who always survey the same site. So perhaps this should be conditional on a setting.

jaxxzer, 2020-02-07:
@hamishwillee great input. @DonLakeFlyer @dogmaphobic does the proposal sound ok to you?

dogmaphobic, 2020-02-07:
I'm not sure I get how you will set the name of the file recorded (there is no API for that). The camera will record using whatever format it does. Photos are geotagged, which would at least tell you where they were taken. Not sure how to handle videos.

dogmaphobic, 2020-02-07:
Ha! I now see this: https://github.com/mavlink/qgroundcontrol/pull/8285/files

That's for local recording only, which I would assume is only used by a fraction of video recording cases. This might lead to confusion as the bulk of "video recording" happens within the camera itself.

jaxxzer, 2020-02-07:
@tcanabrava and I decided that a popup is a bit intrusive to operations as @hamishwillee points out. We can move the text entry field from the general settings page to the camera management widget under the compass. Then, the user can enter the jobsite or not before they hit 'record', no popup and easy to navigate.

> That's for local recording only, which I would assume is only used by a fraction of video recording cases. This might lead to confusion as the bulk of "video recording" happens within the camera itself.

The GCS is the primary (only) recording device for ArduSub, but I know that's a bit of a specialized case with the high-bandwidth link. I think most other ardupilot users with a companion computer use this same method to store the flight video, and high bandwidth/long range wireless links are becoming more and more frequent too. Some security against a 'total loss' is something to consider too when the unexpected happens.

DonLakeFlyer, 2020-02-07:
Seems reasonable. I'd prime the filename with the date/time name. Then they can change it to whatever they want. If you are talking about the name only being a prefix for the filename then that's a whole different story and too complicated I think.

dogmaphobic, 2020-02-07:
It needs to be made clear that this is only related to _local recording_ and it will not set anything if the video is being recorded within the camera.

DonLakeFlyer, 2020-02-07:
> It needs to be made clear that this is only related to _local recording_ and it will not set anything if the video is being recorded within the camera.

Are those two different buttons you press to record video? I agree that if this can't be made clear we shouldn't do it.

booo, 2022-09-15:
As far as I can see it's now possible to change the name of the file. Please reopen if you need more features.

closed 2022-09-15 by booo