mavlink/qgroundcontrol #8252: Allow building QGroundControl with only LGPL Qt modules

realflight1, 2020-01-23:
According to [Qt Licensing](https://doc-snapshots.qt.io/qt5-5.12/licensing.html) and in particular [this list](https://doc-snapshots.qt.io/qt5-5.12/qtmodules.html#gpl-licensed-addons), most of Qt is (also) released under the LGPL license except for a few modules that are GPL / commercial only.

QGroundControl uses mostly LGPL modules. At the moment only MAVLink inspector uses the charts module which is GPL / commercial.

What I am asking for is a CONFIG option or something similar to disable building code that uses non-LGPL modules. This will require changes to qgroundcontrol.pro. Something like this works for the current code base (define QGC_ENABLE_QT_LGPL in user_config.pri or on the command line):

    #-------------------------------------------------------------------------------------
    # MAVLink Inspector
    contains (DEFINES, QGC_ENABLE_MAVLINK_INSPECTOR && !QGC_ENABLE_QT_LGPL) {

Another option would be something like this (easier to understand but more verbose):

    # Enable usage of LGPL-only Qt modules
    contains(CONFIG, QGC_ENABLE_QT_LGPL) {
        # Disable MAVLink inspector as it uses the charts module
        contains (DEFINES, QGC_ENABLE_MAVLINK_INSPECTOR) {
            DEFINES -= QGC_ENABLE_MAVLINK_INSPECTOR
        }
    }

You might remember I started a [discussion](https://discuss.px4.io/t/licensing-question-qgc-qt-open-source/13889) on this.

Thanks!

DonLakeFlyer, 2020-02-07:
> At the moment only MAVLink inspector uses the charts module which is GPL / commercial.

Just FYI: PID tuning also uses charting

dogmaphobic, 2020-02-07:
That means both iOS and Android must also be disabled as they both statically link Qt, no?

realflight1, 2020-02-11:
> > At the moment only MAVLink inspector uses the charts module which is GPL / commercial.
> 
> Just FYI: PID tuning also uses charting

Correct, my mistake. QGC can still be built though, it will just fail when trying to open Tuning with the following message (tested with ArduCopter SITL) -

    qrc:/qml/APMTuningComponentCopter.qml:463:17: Type PIDTuning unavailable
    qrc:/qml/QGroundControl/Controls/PIDTuning.qml:12:1: module "QtCharts" is not installed

> That means both iOS and Android must also be disabled as they both statically link Qt, no?

Regarding Android, QGC already links dynamically with QT. You can unzip the generated APK file and see that it includes the QT .so files. I am not sure about iOS.

ea-r1, 2024-03-06:
Would this be possible ?

closed 2026-10-04 by github-actions[bot]