mavlink/qgroundcontrol #8281: CONFIG+=DISABLE_VIDEOSTREAMING no longer builds

DonLakeFlyer, 2020-02-07:
linking debug/QGroundControl
obj/VideoStreaming.o: In function `int qmlRegisterType<GLVideoItemStub>(char const*, int, int, char const*)':
/home/parallels/Qt/5.12.6/gcc_64/include/QtQml/qqml.h:285: undefined reference to `GLVideoItemStub:Makefile:1929: recipe for target 'debug/QGroundControl' failed
:staticMetaObject'
/home/parallels/Qt/5.12.6/gcc_64/include/QtQml/qqml.h:287: undefined reference to `GLVideoItemStub::staticMetaObject'
obj/VideoStreaming.o: In function `QQmlPrivate::QQmlElement<GLVideoItemStub>::QQmlElement()':
/home/parallels/Qt/5.12.6/gcc_64/include/QtQml/qqmlprivate.h:103: undefined reference to `GLVideoItemStub::GLVideoItemStub()'
obj/VideoStreaming.o: In function `QtPrivate::MetaObjectForType<GLVideoItemStub*, void>::value()':
/home/parallels/Qt/5.12.6/gcc_64/include/QtCore/qmetatype.h:1476: undefined reference to `GLVideoItemStub::staticMetaObject'
obj/VideoStreaming.o: In function `QMetaTypeIdQObject<GLVideoItemStub*, 8>::qt_metatype_id()':
/home/parallels/Qt/5.12.6/gcc_64/include/QtCore/qmetatype.h:1832: undefined reference to `GLVideoItemStub::staticMetaObject'
obj/VideoStreaming.o:(.data.rel.ro._ZTVN11QQmlPrivate11QQmlElementI15GLVideoItemStubEE[_ZTVN11QQmlPrivate11QQmlElementI15GLVideoItemStubEE]+0x10): undefined reference to `GLVideoItemStub::metaObject() const'
obj/VideoStreaming.o:(.data.rel.ro._ZTVN11QQmlPrivate11QQmlElementI15GLVideoItemStubEE[_ZTVN11QQmlPrivate11QQmlElementI15GLVideoItemStubEE]+0x18): undefined reference to `GLVideoItemStub::qt_metacast(char const*)'
obj/VideoStreaming.o:(.data.rel.ro._ZTVN11QQmlPrivate11QQmlElementI15GLVideoItemStubEE[_ZTVN11QQmlPrivate11QQmlElementI15GLVideoItemStubEE]+0x20): undefined reference to `GLVideoItemStub::qt_metacall(QMetaObject::Call, int, void**)'
obj/VideoStreaming.o:(.data.rel.ro._ZTIN11QQmlPrivate11QQmlElementI15GLVideoItemStubEE[_ZTIN11QQmlPrivate11QQmlElementI15GLVideoItemStubEE]+0x10): undefined reference to `typeinfo for GLVideoItemStub'
obj/VideoStreaming.o: In function `QQmlPrivate::QQmlElement<GLVideoItemStub>::~QQmlElement()':
/home/parallels/Qt/5.12.6/gcc_64/include/QtQml/qqmlprivate.h:108: undefined reference to `GLVideoItemStub::~GLVideoItemStub()'
collect2: error: ld returned 1 exit status

DonLakeFlyer, 2020-02-19:
Well that didn't last long. This is broken again:
```
/home/parallels/repos/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:910: error: ‘GstElement’ does not name a type; did you mean ‘qstrlen’?
 GstElement*
 ^~~~~~~~~~
 qstrlen
```

With a pile of others...

andrewvoznytsa, 2020-02-19:
Will fix shortly.

BTW, can we add that build type to CI? Or it will overload system?

DonLakeFlyer, 2020-02-19:
> can we add that build type to CI

The problem is it will increase the length of build time turnaround too much

DonLakeFlyer, 2020-02-20:
It lasted for even a short time this go around!
```
/home/parallels/repos/qgroundcontrol/src/VideoStreaming/VideoStreaming.cc:201: error: ‘debuglevel’ was not declared in this scope
     Q_UNUSED(debuglevel)
              ^
```

andrewvoznytsa, 2020-02-20:
I'll beat PR submitting rate record if I keep failing with that build type so frequently.

closed 2020-02-20 by DonLakeFlyer

AndreySkyFoxSidorov, 2022-08-10:
C:\\Qt\Qt5.15.2\5.15.2\android\include\QtQml\qqml.h:-1: error: error: undefined reference to 'GLVideoItemStub::staticMetaObject'
C:\\Qt\Qt5.15.2\5.15.2\android\include\QtCore\qmetatype.h:-1: error: error: undefined reference to 'GLVideoItemStub::staticMetaObject'
C:\\Qt\Qt5.15.2\5.15.2\android\include\QtCore\qmetatype.h:-1: error: error: undefined reference to 'GLVideoItemStub::staticMetaObject'
C:\\Qt\Qt5.15.2\5.15.2\android\include\QtQml\qqmlprivate.h:139: error: error: undefined reference to 'GLVideoItemStub::GLVideoItemStub()'
C:\\Qt\Qt5.15.2\5.15.2\android\include\QtQml\qqmlprivate.h:144: error: error: undefined reference to 'GLVideoItemStub::~GLVideoItemStub()'
C:\\Qt\Qt5.15.2\5.15.2\android\include\QtQml\qqmlprivate.h:144: error: error: undefined reference to 'GLVideoItemStub::~GLVideoItemStub()'
:-1: error: linker command failed with exit code 1 (use -v to see invocation)
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [Makefile:2601: libQGroundControl_armeabi-v7a.so] Error 1

deadman2000, 2022-08-30:
@AndreySkyFoxSidorov try to clean the project and rebuild. this should help