mavlink/qgroundcontrol #8282: Ubuntu 18.04 builds fail link

DonLakeFlyer, 2020-02-07:
```
linking debug/QGroundControl
//usr/lib/x86_64-linux-gnu/libEGL.so.1: undefined reference to `g_PrlGLXApi'
collect2: error: ld returned 1 exit status
```

Gus can reproduce as well the same thing. No idea why travis CI builds work which also run on 18.04. It seems to be related to gstreamer. If I hack that out I can get a build to link.

DonLakeFlyer, 2020-02-07:
If I hack up the build to make video disable work then the link works fine. So something related to video streaming code.

kenny-chen, 2020-02-12:
Hey DonLakeFlyer -- I had this exact same problem building a project from source this was the only page from Google that had a reference to "g_PrlGLXApi." FYI after many hours I figured out this was a problem with how Parallels is handling graphics (since I see you're using Parallels as well from your link to #8281). Try disabling 3D acceleration, vertical sync, and automatic graphics switching in your VM configuration in Parallels. I'm not sure exactly which setting fixed it, but the build links now. This would probably also explain why it builds fine in Travis for you. Just wanted to pop in and let you know.

DonLakeFlyer, 2020-02-12:
> Try disabling 3D acceleration, vertical sync, and automatic graphics switching in your VM configuration in Parallels.

Thanks for the info. I'd consider that a last resort though. We'd want people to be able to build QGC with an out of the box parallels setup. I think that's a fairly common thing.

andrewvoznytsa, 2020-02-12:
I can't run Parallels so I'd ask someone to do a few simple tests:
1) please run `gst-launch-1.0 filesrc location=~/Documents/QGroundControl/Video/2020-02-11_22.23.03.mkv ! decodebin ! glimagesink` where `~/Documents/QGroundControl/Video/2020-02-11_22.23.03.mkv` could be replaced by path to some video. Preferably - to the video recorded by QGC. Expected behavior - you'll see some video (hopefully no - I'll say that it is gstreamer' issue). 
2) please run 'ldd /usr/lib/x86_64-linux-gnu/libgstgl-1.0.so' and send me output.
3) please run 'ldd QGroundControl' and also send me output (you'll need to replace QGroundControl by actual path to QGroundControl).

DonLakeFlyer, 2020-02-12:
```
parallels@parallels:~/repos/Firmware$ ldd /usr/lib/x86_64-linux-gnu/libgstgl-1.0.so
	linux-vdso.so.1 (0x00007fff180e4000)
	libgstvideo-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libgstvideo-1.0.so.0 (0x00007fd2c65c7000)
	libgstallocators-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libgstallocators-1.0.so.0 (0x00007fd2c63c3000)
	libgmodule-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgmodule-2.0.so.0 (0x00007fd2c61bf000)
	libgstbase-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libgstbase-1.0.so.0 (0x00007fd2c5f4a000)
	libgstreamer-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libgstreamer-1.0.so.0 (0x00007fd2c5c0f000)
	libGL.so.1 => /usr/lib/x86_64-linux-gnu/libGL.so.1 (0x00007fd2c5996000)
	libX11-xcb.so.1 => /usr/lib/x86_64-linux-gnu/libX11-xcb.so.1 (0x00007fd2c5794000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007fd2c545c000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007fd2c5234000)
	libwayland-egl.so.1 => /usr/lib/x86_64-linux-gnu/libwayland-egl.so.1 (0x00007fd2c5032000)
	libwayland-client.so.0 => /usr/lib/x86_64-linux-gnu/libwayland-client.so.0 (0x00007fd2c4e23000)
	libEGL.so.1 => /usr/lib/x86_64-linux-gnu/libEGL.so.1 (0x00007fd2c4c11000)
	libdrm.so.2 => /usr/lib/x86_64-linux-gnu/libdrm.so.2 (0x00007fd2c4a00000)
	libgbm.so.1 => /usr/lib/x86_64-linux-gnu/libgbm.so.1 (0x00007fd2c47fc000)
	libgudev-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libgudev-1.0.so.0 (0x00007fd2c45f2000)
	libgobject-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007fd2c439e000)
	libglib-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007fd2c4087000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fd2c3e68000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fd2c3a77000)
	liborc-0.4.so.0 => /usr/lib/x86_64-linux-gnu/liborc-0.4.so.0 (0x00007fd2c37fb000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fd2c345d000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fd2c3259000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fd2c3051000)
	libPrlDRI.so.1 => /usr/lib/x86_64-linux-gnu/libPrlDRI.so.1 (0x00007fd2c2d62000)
	libPrlWl.so.1 => /usr/lib/x86_64-linux-gnu/libPrlWl.so.1 (0x00007fd2c2b59000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fd2c27d0000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fd2c25b8000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007fd2c23b4000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007fd2c21ae000)
	libffi.so.6 => /usr/lib/x86_64-linux-gnu/libffi.so.6 (0x00007fd2c1fa6000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007fd2c1d88000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007fd2c1b16000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fd2c6ac6000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007fd2c1901000)
parallels@parallels:~/repos/Firmware$ 
```

DonLakeFlyer, 2020-02-12:
This is from latest master tarball build from S3:
```
parallels@parallels:~/Downloads/qgroundcontrol$ ldd QGroundControl 
./QGroundControl: /usr/lib/x86_64-linux-gnu/libQt5Core.so.5: version `Qt_5.12' not found (required by ./QGroundControl)
	linux-vdso.so.1 (0x00007fff1eff3000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f1d22b85000)
	libSDL2-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libSDL2-2.0.so.0 (0x00007f1d22853000)
	libgstgl-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libgstgl-1.0.so.0 (0x00007f1d225ed000)
	libgstvideo-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libgstvideo-1.0.so.0 (0x00007f1d22354000)
	libgstbase-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libgstbase-1.0.so.0 (0x00007f1d220df000)
	libgstreamer-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libgstreamer-1.0.so.0 (0x00007f1d21da4000)
	libgobject-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007f1d21b50000)
	libglib-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f1d21839000)
	libQt5Location.so.5 => not found
	libQt5Quick.so.5 => not found
	libQt5Charts.so.5 => not found
	libQt5Widgets.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x00007f1d20ff2000)
	libQt5Multimedia.so.5 => not found
	libQt5X11Extras.so.5 => not found
	libQt5Gui.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007f1d20889000)
	libQt5Positioning.so.5 => not found
	libQt5Qml.so.5 => not found
	libQt5Network.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Network.so.5 (0x00007f1d204fd000)
	libQt5Sql.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Sql.so.5 (0x00007f1d202b4000)
	libQt5Xml.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Xml.so.5 (0x00007f1d20078000)
	libQt5TextToSpeech.so.5 => not found
	libQt5SerialPort.so.5 => not found
	libQt5Test.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Test.so.5 (0x00007f1d1fe3e000)
	libQt5Core.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007f1d1f6f3000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f1d1f36a000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f1d1efcc000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f1d1edb4000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f1d24aec000)
	libasound.so.2 => /usr/lib/x86_64-linux-gnu/libasound.so.2 (0x00007f1d1eaad000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f1d1e8a9000)
	libpulse.so.0 => /usr/lib/x86_64-linux-gnu/libpulse.so.0 (0x00007f1d1e659000)
	libsndio.so.6.1 => /usr/lib/x86_64-linux-gnu/libsndio.so.6.1 (0x00007f1d1e449000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f1d1e111000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007f1d1deff000)
	libXcursor.so.1 => /usr/lib/x86_64-linux-gnu/libXcursor.so.1 (0x00007f1d1dcf5000)
	libXinerama.so.1 => /usr/lib/x86_64-linux-gnu/libXinerama.so.1 (0x00007f1d1daf2000)
	libXi.so.6 => /usr/lib/x86_64-linux-gnu/libXi.so.6 (0x00007f1d1d8e2000)
	libXrandr.so.2 => /usr/lib/x86_64-linux-gnu/libXrandr.so.2 (0x00007f1d1d6d7000)
	libXss.so.1 => /usr/lib/x86_64-linux-gnu/libXss.so.1 (0x00007f1d1d4d3000)
	libXxf86vm.so.1 => /usr/lib/x86_64-linux-gnu/libXxf86vm.so.1 (0x00007f1d1d2cd000)
	libwayland-egl.so.1 => /usr/lib/x86_64-linux-gnu/libwayland-egl.so.1 (0x00007f1d1d0cb000)
	libwayland-client.so.0 => /usr/lib/x86_64-linux-gnu/libwayland-client.so.0 (0x00007f1d1cebc000)
	libwayland-cursor.so.0 => /usr/lib/x86_64-linux-gnu/libwayland-cursor.so.0 (0x00007f1d1ccb4000)
	libxkbcommon.so.0 => /usr/lib/x86_64-linux-gnu/libxkbcommon.so.0 (0x00007f1d1ca75000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f1d1c856000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f1d1c64e000)
	libgstallocators-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libgstallocators-1.0.so.0 (0x00007f1d1c44a000)
	libgmodule-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgmodule-2.0.so.0 (0x00007f1d1c246000)
	libGL.so.1 => /usr/lib/x86_64-linux-gnu/libGL.so.1 (0x00007f1d1bfcd000)
	libX11-xcb.so.1 => /usr/lib/x86_64-linux-gnu/libX11-xcb.so.1 (0x00007f1d1bdcb000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f1d1bba3000)
	libEGL.so.1 => /usr/lib/x86_64-linux-gnu/libEGL.so.1 (0x00007f1d1b991000)
	libdrm.so.2 => /usr/lib/x86_64-linux-gnu/libdrm.so.2 (0x00007f1d1b780000)
	libgbm.so.1 => /usr/lib/x86_64-linux-gnu/libgbm.so.1 (0x00007f1d1b57c000)
	libgudev-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libgudev-1.0.so.0 (0x00007f1d1b372000)
	liborc-0.4.so.0 => /usr/lib/x86_64-linux-gnu/liborc-0.4.so.0 (0x00007f1d1b0f6000)
	libffi.so.6 => /usr/lib/x86_64-linux-gnu/libffi.so.6 (0x00007f1d1aeee000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f1d1ac7c000)
	libpng16.so.16 => /usr/lib/x86_64-linux-gnu/libpng16.so.16 (0x00007f1d1aa4a000)
	libharfbuzz.so.0 => /usr/lib/x86_64-linux-gnu/libharfbuzz.so.0 (0x00007f1d1a7ac000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f1d1a58f000)
	libicui18n.so.60 => /usr/lib/x86_64-linux-gnu/libicui18n.so.60 (0x00007f1d1a0ee000)
	libicuuc.so.60 => /usr/lib/x86_64-linux-gnu/libicuuc.so.60 (0x00007f1d19d37000)
	libdouble-conversion.so.1 => /usr/lib/x86_64-linux-gnu/libdouble-conversion.so.1 (0x00007f1d19b26000)
	libpulsecommon-11.1.so => /usr/lib/x86_64-linux-gnu/pulseaudio/libpulsecommon-11.1.so (0x00007f1d198a8000)
[... 3 lines cut ...]
	libXfixes.so.3 => /usr/lib/x86_64-linux-gnu/libXfixes.so.3 (0x00007f1d19036000)
	libPrlDRI.so.1 => /usr/lib/x86_64-linux-gnu/libPrlDRI.so.1 (0x00007f1d18d47000)
	libPrlWl.so.1 => /usr/lib/x86_64-linux-gnu/libPrlWl.so.1 (0x00007f1d18b3e000)
[... cut ...]
	libFLAC.so.8 => /usr/lib/x86_64-linux-gnu/libFLAC.so.8 (0x00007f1d14f90000)
	libogg.so.0 => /usr/lib/x86_64-linux-gnu/libogg.so.0 (0x00007f1d14d87000)
	libvorbis.so.0 => /usr/lib/x86_64-linux-gnu/libvorbis.so.0 (0x00007f1d14b5c000)
	libvorbisenc.so.2 => /usr/lib/x86_64-linux-gnu/libvorbisenc.so.2 (0x00007f1d148b3000)
	libresolv.so.2 => /lib/x86_64-linux-gnu/libresolv.so.2 (0x00007f1d14698000)
	libgpg-error.so.0 => /lib/x86_64-linux-gnu/libgpg-error.so.0 (0x00007f1d14483000)
parallels@parallels:~/Downloads/qgroundcontrol$ 
```

DonLakeFlyer, 2020-02-12:
I can't do anything for #1 since I don't have a vehicle with video support.

andrewvoznytsa, 2020-02-12:
https://drive.google.com/open?id=11yQMVKTlSEl8eABvWmZ3HrkKR1Kem0if - it is my 2020-02-11_22.23.03.mkv

dogmaphobic, 2020-02-13:
For the #1:

```
~$ gst-launch-1.0 filesrc location=./2020-02-11_22.23.03.mkv ! decodebin ! glimagesink
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Got context from element 'sink': gst.gl.GLDisplay=context, gst.gl.GLDisplay=(GstGLDisplay)"\(GstGLDisplayX11\)\ gldisplayx11-0";
Redistribute latency...
Redistribute latency...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Got EOS from element "pipeline0".
Execution ended after 0:00:08.818923558
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```
The video plays just fine (ball moving around)

[image]

dogmaphobic, 2020-02-13:
But I can't link it either:
```
linking debug/QGroundControl
//usr/lib/x86_64-linux-gnu/libEGL.so.1: undefined reference to `g_PrlGLXApi'
collect2: error: ld returned 1 exit status
```

andrewvoznytsa, 2020-02-13:
OK, for me it sounds like issue with pkg-config for some libEGL or some other GL related stuff that is part of your system (Parelles GL implementation).

I believe that g_PrlGLXApi popped up after switching to qmlglsink. This switch added gstreamer-gl-1.0 to PKGCONFIG (now it is `PKGCONFIG   += gstreamer-1.0  gstreamer-video-1.0 gstreamer-gl-1.0`) and sounds that gstreamer-gl-1.0 (or one of its dependencies) is missing some lib. I suppose that if you add `LIBS += -lPrlDRI -lPrlWl` then it will solve your build issue. Could you please check?

And I'm right assuming that you can install daily build with enabled video and it works without issues?

andrewvoznytsa, 2020-02-13:
It is how your LInux section of VideoStreaming.pri could look like:

`LinuxBuild {
    QT += x11extras waylandclient
    CONFIG += link_pkgconfig
    packagesExist(gstreamer-1.0) {
        PKGCONFIG   += gstreamer-1.0  gstreamer-video-1.0 gstreamer-gl-1.0
        LIBS += -lPrlDRI -lPrlWl
        CONFIG      += VideoEnabled
    }
} else:MacBuild {
`

dogmaphobic, 2020-03-07:
Need to add the egl package. It now builds:
```
PKGCONFIG   += gstreamer-1.0 gstreamer-video-1.0 gstreamer-gl-1.0 egl
```

Need to test this on a "normal" Ubuntu system.

andrewvoznytsa, 2020-03-08:
Ubuntu 18.04 `pkg-config --libs egl` is `-lEGL`

Works for me (linking succeed).

bys1123, 2020-03-10:
I got 
```
/tmp/.mount_FThcTb/QGroundControl: error while loading shared libraries: libgstgl-1.0.so.0: cannot open shared object file: No such file or directory
```
on unbuntu 16.04
when I run `./QGroundControl-4.0.2.AppImage`

closed 2020-03-17 by andrewvoznytsa