mavlink/qgroundcontrol #8283: No RTSP video in Beta Groundcontrol

Oeble1966, 2020-02-07:
----

## Expected Behavior
Like in official / stable version , RTSP video should be a choice.

## Current Behavior
Only local machine camera;s can be chosen ( front , back ect)

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10 and Android 8
- QGC Version::daily's
- QGC build: daily
- Flight Controller: [e.g., Pixhawk 
- Autopilot (with version):Ardupilot Copter 4.0

## Detailed Description
I use the eachine ROTG to see FPV video on tablet , phone , or windows.. All works fine in Stable , in Daily ( that i need for extra options ) there is no possibility to choose RTSP video ..only the hardware in the devices.

andrewvoznytsa, 2020-02-12:
@Oeble1966 what you see could be related to https://github.com/mavlink/qgroundcontrol/pull/8318 if your QGC was running on Windows 10. With https://github.com/mavlink/qgroundcontrol/pull/8318 I've RTSP works on Windows for me.

If you were also running QGC on Android 8 - please confirm - works for me on Android 6.

Oeble1966, 2020-02-12:
@andrewvoznytsa
Hi , i'm running Qgroundcontrol on Android * (tablet) and Android 10 (phone) .Latest stable is Okay , Beta build has no RTSP or other options than Front / Back camera. See attached..

[image]
[image]
[image]

dogmaphobic, 2020-02-12:
The one that only has UVC devices (daily) was built with no GStreamer support (all GStreamer options are missing).

DonLakeFlyer, 2020-02-12:
The 32 bit build is building correctly with gstreamer but the 64 bit build is not building with gstreamer. You can see it in the travis output:
```
Project MESSAGE: Qt version 5.12.6
1413Project MESSAGE: Android Arm 64 bit build
1414/usr/bin/ccache
1415Project MESSAGE: Found ccache, enabling
1416Project MESSAGE: QGroundControl Development HEAD:afe20ca35 2020-02-11 19:56:47 +0000
1417Project MESSAGE: Release flavor
1418Project MESSAGE: Using Default QtLocation headers
1419Project MESSAGE: Skipping support for Pairing
1420Project MESSAGE: Using MAVLink dialect 'ardupilotmega'.
1421Project MESSAGE: ANDROID_EXTRA_LIBS
1422Project MESSAGE: arm64-v8a
1423Project MESSAGE: /home/travis/build/mavlink/qgroundcontrol/libs/OpenSSL/android_openssl/arm64/libcrypto.so /home/travis/build/mavlink/qgroundcontrol/libs/OpenSSL/android_openssl/arm64/libssl.so
1424Project MESSAGE: Skipping support for Zeroconf (unsupported platform)
1425Project MESSAGE: Looking for Airmap in folder "/home/travis/build/mavlink/qgroundcontrol/libs/airmapd", variant: "Qt.5.12"
1426Project MESSAGE: Skipping support for Airmap (unsupported platform)
1427Project MESSAGE: Skipping support for video streaming (GStreamer libraries not installed)
1428Project MESSAGE: Installation instructions here: https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoStreaming/README.md
1429Project MESSAGE: Adding Serial Java Classes
1430Project MESSAGE: This project is using private headers and will therefore be tied to this specific Qt module build version.
1431Project MESSAGE: Running this project against other versions of the Qt modules may crash at any arbitrary point.
1432Project MESSAGE: This is not a bug, but a result of using Qt internals. You have been warned!
1433
```

DonLakeFlyer, 2020-02-12:
The .pri file check is this:
```
    #- gstreamer assumed to be installed in $$PWD/../../gstreamer-1.0-android-universal-1.14.4/***
    contains(QT_ARCH, arm) {
        GST_ROOT = $$PWD/../../gstreamer-1.0-android-universal-1.14.4/armv7
    } else:contains(QT_ARCH, arm64) {
        GST_ROOT = $$PWD/../../gstreamer-1.0-android-universal-1.14.4/arm64
    } else {
        GST_ROOT = $$PWD/../../gstreamer-1.0-android-universal-1.14.4/x86
    }
```

Could travis be installing the wrong thing? Here is travis:
```
  apt:
    packages:
    - speech-dispatcher
    - libgstreamer-plugins-base1.0-dev
    - libgstreamer1.0-0:amd64
    - libgstreamer1.0-dev
    - libudev-dev
    - wget
```

andrewvoznytsa, 2020-02-12:
I think you are mixing build (host) system and target system options.

`GST_ROOT = $$PWD/../../gstreamer-1.0-android-universal-1.14.4/arm64` is for target system (android64) while as
`apt:
    packages:`
from travis is for build (host) system.

If travis is building for Android then it should download https://gstreamer.freedesktop.org/data/pkg/android/1.14.4/gstreamer-1.0-android-universal-1.14.4.tar.bz2 (see https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoStreaming/README.md, Android section) and unpack that. This archive contains arm/arm64/armv7/x86/x86_64 gstreamer builds for Android - it is all what you need from video side to enable video support on Android.

I just tested QtCreator/Android/amr64 building on my side - all good.

Looking on travis config I see that it is downloading some other files (https://s3-us-west-2.amazonaws.com/qgroundcontrol/dependencies/Qt5.12.6-android_arm64_v8a-min.tar.bz2 for example) - not sure if they are good/compatible with current video code.

I'll look into that.

DonLakeFlyer, 2020-02-12:
Sorry that was the wrong section. This is what CI does:
```
  # android dependencies: qt, gstreamer, android-ndk
  - if [ "${SPEC}" = "android-clang" ]; then
        wget --quiet https://s3-us-west-2.amazonaws.com/qgroundcontrol/dependencies/gstreamer-1.0-android-universal-1.14.4.tar.bz2 &&
        tar jxf gstreamer-1.0-android-universal-1.14.4.tar.bz2 -C ${TRAVIS_BUILD_DIR} &&
        wget --quiet https://dl.google.com/android/repository/android-ndk-r20-linux-x86_64.zip &&
        unzip android-ndk-r20-linux-x86_64.zip > /dev/null &&
        export ANDROID_NDK_ROOT=`pwd`/android-ndk-r20 &&
        export ANDROID_SDK_ROOT=/usr/local/android-sdk &&
        export PATH=`pwd`/android-ndk-r20:$PATH
        ;
    fi
```

andrewvoznytsa, 2020-02-12:
I'm downloading already https://s3-us-west-2.amazonaws.com/qgroundcontrol/dependencies/gstreamer-1.0-android-universal-1.14.4.tar.bz2 - will compare with this one which I usually use (https://gstreamer.freedesktop.org/data/pkg/android/1.14.4/gstreamer-1.0-android-universal-1.14.4.tar.bz2). Then will decide what is next.

DonLakeFlyer, 2020-02-12:
Thanks @andrewvoznytsa. I'm not sure who set up the original travis+S3 stuff. Maybe Gus. So not sure whether that file matches what you need. If you need me to push something different to S3 let me know and I can do that.

dogmaphobic, 2020-02-12:
> I'm not sure who set up the original travis+S3 stuff. Maybe Gus.

I'm 100% sure it wasn't me as I don't even now how to connect to S3 :P

andrewvoznytsa, 2020-02-15:
@DonLakeFlyer to fix that issue you need to replace https://s3-us-west-2.amazonaws.com/qgroundcontrol/dependencies/gstreamer-1.0-android-universal-1.14.4.tar.bz2 (armv7/x86) by https://gstreamer.freedesktop.org/data/pkg/android/1.14.4/gstreamer-1.0-android-universal-1.14.4.tar.bz2 (arm/arm64/armv7/x86/x86_64).

To optimize S3 download time (which is huge for me) you can remove some gstreamer platforms but I think we'll end up in the same story shortly, as someone adds new platform support. Or we can try to split file per architecture and tweak travis to download only these builds which we really need.

DonLakeFlyer, 2020-02-20:
This should be fixed now

closed 2020-02-20 by DonLakeFlyer

Oeble1966, 2020-02-20:
Your all The Best !... Works !

DonLakeFlyer, 2020-02-20:
Awesome thanks for verifying