mavlink/qgroundcontrol #8291: Follow Me settings garbled after setting FOLL_ALT_TYPE to absolute

twistedwings, 2020-02-08:
[image]

[image]

[image]

ArduCopter 4.0.1
[image]

DonLakeFlyer, 2020-02-08:
@twistedwings This is because QGC currently only supports relative. Can you explain what the difference is with the two FOLL_ALT_TYPE modes? I don't see how they change anything with respect to the offsets. If you set to absolute which works differently?

twistedwings, 2020-02-08:
In absolute mode the uav follows at fixed height.
In relative mode it follows relative to the leaders height.
If the leader climbs a mountain the uav will also ascend in relative mode,
in absolute mode it will crash against the mountain.

DonLakeFlyer, 2020-02-08:
Duh, Why couldn't I understand that! That mean the fix I made should work fine.

DonLakeFlyer, 2020-02-08:
Daily builds which spit out in an hour or so should have the fix. Can you verify?

closed 2020-02-08 by DonLakeFlyer

twistedwings, 2020-02-09:
verified in both windows and android version.
Thank you for the fix.

DonLakeFlyer, 2020-02-09:
@twistedwings Thanks for beating on this new feature to wring it out