mavlink/qgroundcontrol #8293: Graphical bugs in Mode and Arm drop down selectors

Antiheavy, 2020-02-08:
Recent daily build, Windows 10.

- drop down menus cut off words for flight modes and arm/disarm
- white squares (maybe map tiles?) pop on the screen when clicking around the drop downs

Video here shows this issue:
https://youtu.be/e2_y6RrPskA

DonLakeFlyer, 2020-02-08:
@dogmaphobic Can you look a the white sqaure thing?

DonLakeFlyer, 2020-02-08:
White black possibly related to #7976

dogmaphobic, 2020-02-10:
> White black possibly related to #7976

It looks like the same bug. This time caused by using a ComboBox from QtQuick.Controls 2.x. We worked around the Menu one by forcing the use of Menus from QtQuick.Controls 1.x. That won't be so simple with the ComboBox. It's highly customized using QtQuick.Controls 2.x functions. 

This occurs when these items are overlaid on "Maps".

DonLakeFlyer, 2020-02-10:
We're going to need to figure this one out for real likely through the Qt folks which is going to be a pain. We can't move things back to control 1.

Antiheavy, 2020-02-16:
The white map squares popping in and out thing happens when working through the Checklist too.

closed 2020-02-16 by dogmaphobic