mavlink/qgroundcontrol #8296: Create Plan: Survey mission missing key steps

Antiheavy, 2020-02-09:
The new Create Plan function goes a long way to help people figure out mission planning on their own.  However, the Survey (and others) plan wizards miss some key steps that requires the user to go back and adjust the Survey (and others) every time.  I think a lot of this could be avoided by letting the user choose a Preset (if they have any) and rotate the transects right in the Wizard.

Here is a mock-up of what I am thinking:
[image]

This becomes more than just creating the Polygon so maybe the button becomes [Done Survey], not sure...

DonLakeFlyer, 2020-02-11:
I don't see that as being "Pre-Wizard" complete. To me it's just a change to show the Presets tab first if there are any presets. The presets tab already has transect rotation in it. Why isn't that good enough?

Antiheavy, 2020-02-11:
It's worth a discussion.  

When I'm using a Survey Wizard, I need to do that following every single time:
1) define the polygon
2) choose one of my preset survey types (I feel like there should always be a "custom" option in the presets drop-down that expands the pane to show all the adjustable items, but that is probably a separate discussion),
3) adjust the direction (based on geometry and wind). 

The above 3 items make me have to click on Survey after I've completed the Landing Pattern to go back and change everything around.  It's additionally confusing because I'm presented with all the details of the Landing Pattern once it is completed, and those are items I almost never change.  It's a disjointed thought process and not intuitive for users.

I don't need the other items in the Presets pane at this stage.  When using a Wizard I don't need buttons for [Apply Preset], [Delete Preset], or [Save Settings as New Preset].  I also don't necessarily need the statistics right away.

DonLakeFlyer, 2020-02-11:
Ok, I understand now. I wasn't thinking about the wizard moving you to the next item. And then having to come back.

closed 2020-02-14 by DonLakeFlyer