mavlink/qgroundcontrol #8304: MESSAGE_INTERVAL implementation

cooldemo, 2020-02-10:
Ardupilot since V4 has implemented different approach to selecting telemetry stream rates that allows for greater granularity. This allows to lower the amount of telemetry that is generated and needs to be transferred to the GCS.
It would be nice if this was also implemented in QGC along with the graphical configuration tool to select interval for every telemetry message.
The change involves transmitting command:
MAV_CMD_SET_MESSAGE_INTERVAL
instead of
REQUEST_DATA_STREAM

DonLakeFlyer, 2020-02-10:
QGC supports adjust the stream rates to what you want:
[image]

Adjusting on a message by message basis is sort of interesting but not something I see happening anytime soon.

cooldemo, 2020-02-10:
I haven't seen this settings on Android and this only uses REQUEST_DATA_STREAM command. You have only 7 settings to adjust and telemetry rate was quite high up to 30kbps with all settings on 1Hz.
MAV_CMD_SET_MESSAGE_INTERVAL allows much finer granularity and intervals between messages up to 60 seconds.

DonLakeFlyer, 2020-02-10:
> I haven't seen this settings on Android and this only uses REQUEST_DATA_STREAM command.

They are available in latest daily builds.

cooldemo, 2020-02-11:
Is there any plan to remove deprecated REQUEST_DATA_STREAM interface and switch to MAV_CMD_SET_MESSAGE_INTERVAL ?
This would enable lowering telemetry rates to minimum, possibly supporting slower interfaces over longer range.

DonLakeFlyer, 2020-02-11:
> Is there any plan to remove deprecated REQUEST_DATA_STREAM interface and switch to MAV_CMD_SET_MESSAGE_INTERVAL ?

No.

> This would enable lowering telemetry rates to minimum, possibly supporting slower interfaces over longer range.

They way people are doing that is to turn off GCS control of streaming and control from vehicle side through parameters.

cooldemo, 2020-02-11:
> No.
>
>
> They way people are doing that is to turn off GCS control of streaming and control from vehicle side through parameters.

Yes, but this allows setting of only seven parameters and only down to 1Hz. It is still producing a lot of telemetry.
Link to Ardupilot issue where I mentioned it: https://github.com/ArduPilot/ardupilot/issues/13530

DonLakeFlyer, 2020-02-11:
I get what it will support. It's just not something that will bubble to the top of the work list.

closed 2026-10-04 by github-actions[bot]