mavlink/qgroundcontrol #8305: map.toCoordinate

EsbenFR, 2020-02-10:
Hi not a bug, just a question on how is it possible from a custom build to use the function 
`map.toCoordinate(Qt.point(mouse.x, mouse.y), false /* clipToViewPort */)`

anyone have an example

Thanks

Esben

DonLakeFlyer, 2020-02-10:
That is a function on the Qml Map object. If you have access to that object you just use it. Don't understand the question.

closed 2020-02-10 by DonLakeFlyer

EsbenFR, 2020-02-11:
My Question is then, how do I get access to the Qml Map Object
for example in MapScale.qml 

`property var    mapControl                      ///< Map control for which this scale control is being used`

how do i assign the mapcontrol a Qml Map object?

Thanks

Esben

DonLakeFlyer, 2020-02-11:
That all depends on where you are in the code. It might not be possible. You need to trace up the Qml hierarchy to see where it us.