mavlink/qgroundcontrol #8309: Daily build crashes when opening a video stream

kshahar, 2020-02-11:
When selecting a video stream (e.g. "UDP h.264 Video Stream", "MPEG-TS") QGC crashes immediately. If a UDP stream is defined in the .ini file the software crashes on startup. 

## System Information
- Windows 10 Home
- QGC daily
- ArduCopter V4.0.2-dev (98adeafa) - SITL
- Running on Asus ROG GL502V laptop with NVidia GTX 1060

## Detailed Description
I had no such issues when building from source about two weeks ago. Can this be related to #8266? Looks like an OpenGL issue.

The issue is probably not related to the actual video stream - I was not streaming anything while testing. There are no other applications listening on the UDP port. I haven't tested TCP but I suspect the behavior is the same.

After building QGC from source and starting the debug version with GST_DEBUG=3 I get the following output:

```
** (QGroundControl.exe:13264): CRITICAL **: 12:02:41.769: gst_qt_get_gl_wrapcontext: assertion 'wgl_device != NULL' failed
0:00:07.448617000 13264 0000023BEE08E080 WARN               glcontext gstglcontext.c:1240:gst_gl_context_create_thread:<glcontextwgl0> Failed to create context
0:00:07.448717600 13264 0000023BEE0497B0 ERROR             qtglwidget qtitem.cc:382:QtGLVideoItemInterface::initWinSys: failed to share contexts through wglShareLists 0xaa
0:00:07.448778400 13264 0000023BEE0497B0 WARN                  qtsink gstqtsink.cc:309:gst_qt_sink_change_state:<qmlglsink> error: Could not initialize window system
Could not initialize window system
```

The crash is in [VideoReceiver.cc:622](https://github.com/mavlink/qgroundcontrol/blob/b45b853087336664ceef3a368ff65680e382f286/src/VideoStreaming/VideoReceiver.cc#L622). The crash dump in VS 2017 looks like this:

```
 	libgstreamer-1.0-0.dll!000000006dc56f14()	Unknown
 	libgstreamer-1.0-0.dll!000000006dc77286()	Unknown
 	libgstreamer-1.0-0.dll!000000006dc77c9c()	Unknown
>	QGroundControl.exe!VideoReceiver::start() Line 622	C++
 	QGroundControl.exe!VideoReceiver::_updateTimer() Line 1191	C++
 	QGroundControl.exe!QtPrivate::FunctorCall<QtPrivate::IndexesList<>,QtPrivate::List<>,void,void (__cdecl VideoReceiver::*)(void)>::call(void(VideoReceiver::*)() f, VideoReceiver * o, void * * arg) Line 152	C++
 	QGroundControl.exe!QtPrivate::FunctionPointer<void (__cdecl VideoReceiver::*)(void)>::call<QtPrivate::List<>,void>(void(VideoReceiver::*)() f, VideoReceiver * o, void * * arg) Line 186	C++
 	QGroundControl.exe!QtPrivate::QSlotObject<void (__cdecl VideoReceiver::*)(void),QtPrivate::List<>,void>::impl(int which, QtPrivate::QSlotObjectBase * this_, QObject * r, void * * a, bool * ret) Line 415	C++
 	[External Code]	
 	QGroundControl.exe!main(int argc, char * * argv) Line 395	C++
 	[External Code]	
```
The exception is -

```
Unhandled exception at 0x000000006DC56F14 (libgstreamer-1.0-0.dll) in QGroundControl.exe.23180.dmp: 0xC0000005: Access violation reading location 0xFFFFFFFFFFFFFFFF. occurred
```

## Log Files and Screenshots
See attached for a more detailed log file with GST_DEBUG=4 and  --logging:full (although I did remove APM-related logs manually).
[QGC_VideoReceiver_crash.log](https://github.com/mavlink/qgroundcontrol/files/4187233/QGC_VideoReceiver_crash.log)

DonLakeFlyer, 2020-02-11:
@dogmaphobic I'm assigning all of these to you since it the closest I can get to the right person. Not sure if it's possible to get andrey in the system so we can assign to him.

kshahar, 2020-02-12:
I am no longer experiencing this issue after adding HAVE_QT_QPA_HEADER to [qmlglsink.pri:10](https://github.com/mavlink/qgroundcontrol/blob/master/qmlglsink.pri#L10):

```
} else:WindowsBuild {
    DEFINES += HAVE_QT_WIN32 HAVE_QT_QPA_HEADER
    LIBS += opengl32.lib user32.lib
```

It looks like [qtitem.cpp](https://github.com/mavlink/gst-plugins-good/blob/9d782fad9dc0384ba86ecae64511c193f6149f93/ext/qt/qtitem.cc#L304) needs this define (otherwise wgl_device is null) and it is included in the [plugin .pro file](https://github.com/mavlink/gst-plugins-good/blob/9d782fad9dc0384ba86ecae64511c193f6149f93/ext/qt/qtplugin.pro#L21) as well.

dogmaphobic, 2020-02-12:
@andrewvoznytsa 

See above ^^^. This seems promising.

andrewvoznytsa, 2020-02-12:
I'm testing that - 30 mins.

closed 2020-02-15 by DonLakeFlyer