mavlink/qgroundcontrol #8316: Cannot connect to flight controller

dakejahl, 2020-02-12:
I am using Qt 5.12.5 on Windows 10 compiling with MSVC 2017 64bit

**Issue**
Using a local build, the vehicle never "connects", instead I get the message "Your vehicle is not responding. If this continues...etc". Daily builds do work however.

I added some `qDebug()` statements and discovered that QSerialPort **readyRead** signal is never emitted, and thus the **bytesReceived** signal is never emitted and thus the logic in LinkManager.cc fails. A quick google search revealed some known issues.
https://forum.qt.io/topic/107795/qt-5-12-5-seems-not-emit-serialport-readyread-event

dakejahl, 2020-02-12:
The problem is fixed by updating to Qt 5.12.7

closed 2020-02-12 by DonLakeFlyer

dakejahl, 2020-07-25:
FYI The docs still need to be updated to reflect this issue. I've been using 5.12.7 for Windows/Linux/Android, so it might be OK to just bump the version by 1 for all platforms. Someone on Mac would need to verify this is fine.

DonLakeFlyer, 2020-07-25:
This was in 5.12.5 but I thought it was fixed in 5.12.6 which is the current spec'ed version to use.

dakejahl, 2020-07-25:
Ahh okay, I misremembered the version number, I thought it was an issue with 6... it was still in the back of my mind as being "unresolved". Thanks!

Duaa0901, 2021-07-13:
Hello, i have the same issue as yours, have the issue resolved when you have upgrade the QT?