mavlink/qgroundcontrol #8324: not displaying "ADS-B vehicles" on map

kaesesahnetorte, 2020-02-13:
## Expected Behavior
ADS-B vehicle messages sent to QGC via MAVLink are displayed in "MAVLink inspector" as well as on the map

## Current Behavior
ADS-B vehicle messages sent to QGC via mavlink are displayed in "MAVLink inspector" but NOT on the map

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Start QGC
2. Send ADS-B vehice messages to QGC via MAVLink
3. check in "MAVLink inspector" if they were received by QGC
4. check map for aircraft symbols

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10
- QGC Version: Development master b45b85308 from 10.02.2020
or alternatively 29c19c0804103b949c9ad358e7cb4cb7c4ed55bb
                        (stable v3.5.6 merged back into master)
- QGC build: self-built from source
- Flight Controller: Pixhawk 4
- Autopilot: PX4 v0.8

## Detailed Description
I am not experiencing this Problem in QGC v3.5.6 from the installer:
Sending ADS-B vehicle messages will have the planes show up on the map as little grey aircraft icons.
However, when using self-built versions of QGC these aircraft will not show up on the map, although I can see in the MAVLink inspector the messages were received successfully.

## Log Files and Screenshots
[image]
[image]

DonLakeFlyer, 2020-02-13:
> I am not experiencing this Problem in QGC v3.5.6 from the installer:

Does a daily build from the installer work? If so, I can't really help with custom builds. You'll likely have to debug it through yourself to figure it out.

kuiskas, 2020-02-27:
Hi, 
I am experiencing the same issue. I tried all the versions from 3.5.3 to 4.0.0 in either custom build as from the installer. Versions 3.5.3 until 3.5.6 are working, the new 4.0.0 and the daily build are not working. At least for me. 
I have been looking at the code but unfortunately I am not that fit in order to find the error. Little help will be appreciated. 
tested on Windows and linux Linux with Pixhawk 4, PX4 ver. 1.10.1 and ADS-B Receiver Aero from Aerobits.

kaesesahnetorte, 2020-02-27:
I also tried the daily build via installer now and the problem persisted.
Using
Development HEAD:988cd6a9e
from 26.02.2020

The message is shown in the "MavLink Inspector" (as shown in the screenshot 1) but not plotted on the map (how it is supposed to look like - but doesn't - is shown on screenshot 2)

DonLakeFlyer, 2020-02-27:
Can you upload a telemetry log somewhere and then link to it here so I can replay what you are seeing?

kaesesahnetorte, 2020-02-28:
Sure, here is a log using
Development HEAD:988cd6a9e
from 26.02.2020

https://www.file-upload.net/download-13922930/2020-02-2813-26-55.tlog.html

closed 2020-03-01 by DonLakeFlyer

ajinkyaavy, 2020-03-09:
> Hi,
> I am experiencing the same issue. I tried all the versions from 3.5.3 to 4.0.0 in either custom build as from the installer. Versions 3.5.3 until 3.5.6 are working, the new 4.0.0 and the daily [...]

I am using the Aerobits AERO as well. It works on MissionPlanner for me. However, nothing shows up when I try it on QGCS. I shall now check version 3.5.6
Do you have an update on this?

kuiskas, 2020-03-09:
I test it last week. They did fix it on mavlink:Stable_V4.0. Apparently it was just a small bug under 
src/Vehicle/Vehicle.cc 
 line 3881 vehicleInfo.location.setLongitude(adsbVehicleMsg.lon / 1e7);
 line 3882 vehicleInfo.availableFlags |= ADSBVehicle::LocationAvailable;

ajinkyaavy, 2020-03-09:
> I test it last week. They did fix it on mavlink:Stable_V4.0. Apparently it was just a small bug under
> src/Vehicle/Vehicle.cc
> line 3881 vehicleInfo.location.setLongitude(adsbVehicleMsg.lon / 1e7);
> line 3882 vehicleInfo.availableFlags |= ADSBVehicle::LocationAvailable;

Can you share the parameter settings that you used? Or if possible, can you please confirm whether you followed something similar to [this](https://docs.px4.io/master/en/advanced_features/traffic_avoidance_adsb.html)?

kuiskas, 2020-03-09:
just set mavlink to a port to for exemple Telem2. Enable mavlink fordwarding and set the baudrate to **115200bauds**. I do not thing is something else needed.
If a ADSB vehicle is in your proximity you will start to receive ADSB message's and they will be display on your map.

drfrankencopter, 2021-12-01:
I realize that this issue was closed in 2020, but I'm experiencing the same thing with the version I just downloaded from the Stable 4.0 page: 'Development: (hex hash) HEAD 2021-08-31

I've written a custom application that sends MavLink to QGC, and have written up a test function to send ADS-B Vehicle messages to QGC via MavLink UDP (same connection as for all Autopilot messages). I can see my Test ADS-B message in the Mav Link Inspector so I know it's being received...but I don't see any aircraft icon. The Lat/Lon should have it right on the ramp beside my ownship location.

I've attached screenshots, and a log file to this message...

[AppLog.txt](https://github.com/mavlink/qgroundcontrol/files/7636304/AppLog.txt)
[image]
[image]
'.

DonLakeFlyer, 2021-12-01:
USe a daily build instead

drfrankencopter, 2021-12-01:
The issue has been resolved. The problem was between the chair and the monitor! I wasn't passing valid flags (see it's zero in my screen cap) so QGC was filtering it out. Learned this by inspecting vehicle.cc and looking at how the ADS_B vehicle message was handled. I have the flags set, and see traffic now.

Thx!

drfrankencopter, 2021-12-01:
The issue I am running into now is that the Callsign is not being displayed. I have checked the Callsign Valid flag, and there is a callsign of TEST visible in the MavLink Inspector...but I don't see the callsign showing in QGC. I do, however see the altitude, and heading/track as being shown correctly.

I'm browsing the src and trying to get to the bottom of it...