mavlink/qgroundcontrol #8325: Log Download timeouts

DonLakeFlyer, 2020-02-13:
From @fredowski:

 I tested this with the ESP8266 wifi module running firmware version 1.2.2 and Arduplane 4.0.3. The behavior for the log download is

The download starts and reports 20kB/s rate
After approx. 60kB the download is stuck. QGC reports permantly reduced rates. No data is received.
After 8s to 20s the download continues for another maybe 40kB.
After two minutes I have received around in total 300kB.
I have not tested again the USB download. Do you have an idea where the log download protocol is described? Is this maybe a packet queuing problem where the esp stores many packets but cannot forward them and finally dumps the packets?

Moved from #5950

DonLakeFlyer, 2020-02-13:
> Is this maybe a packet queuing problem where the esp stores many packets but cannot forward them and finally dumps the packets?

Hard to say. ArduPilot implements differently if I remember correctly with a sort of burst output of packets. The protocol is specified through the LOG_* mavlink messages: https://mavlink.io/en/messages/common.html.

Let me try some testing of dropped packets against ArduPilot implementation so I can remember how it works.

Can you do your test again turning LogDownloadLog as described here and then attach the Console Log output?

DonLakeFlyer, 2020-02-13:
@fredowski Did you also say somewhere that this same hardware setup and Mission Planner will download logs fine?

DonLakeFlyer, 2020-02-13:
> Let me try some testing of dropped packets against ArduPilot implementation so I can remember how it works.

I tried forcing huge packet loss over a SITL connection and the download still continued to work. So as far I can tell it doesn't look like q QGC side problem.

fredowski, 2020-02-14:
I check with Ivo who tested with Mission Planner but as far as I remember it was also not seamless. Can you tell me how to activate the "LogDownloadLog"? You mentioned "here" in a message above but I guess that was supposed to be a link?

DonLakeFlyer, 2020-02-14:
Oops, sorry: https://docs.qgroundcontrol.com/en/SettingsView/console_logging.html

DonLakeFlyer, 2020-02-21:
From the forums I can see other folks using the new code and it's working.

closed 2020-02-21 by DonLakeFlyer

fredowski, 2020-02-23:
Hi, I tested the log download transfer with your daily build version with a PixRacer running Arduplane 4.0.3 and the ESP Wifi version 1.2.2. The Wifimodule is configured as AP on Channel 6. There is some traffic here in my environment. The rate indicator in QGC reports around 20kB/s. I could transfer 96,8 MB in 2h and 32 minutes.

**This gives an average real log transfer rate of 10.6 kB/s.** 

I noted some intermediate values also and the transfer rate is pretty constant. Looking at the LogDownloadLog from QGC I can see normal retrycounts of up to 4 but I also see 33, 15, 120, 36, 13 by just scrolling through the log. During the transfer I had one occurence of complete "Communication lost" but that was recovered after about 2 minutes. On a second file I have a lost communication with retrycount increasing to 284. 

Thanks for the patch! This makes a download possible when you do not have access to the physical SD Card.

DonLakeFlyer, 2020-02-24:
Thanks for testing @fredowski