mavlink/qgroundcontrol #8327: Fix libicu thing

DonLakeFlyer, 2020-02-13:
See #8219

DonLakeFlyer, 2020-02-13:
@tcanabrava This is to track the libicu thing which needs to happen before stable goes out which is happening soonish.

AJRepo, 2020-10-21:
I've been testing compilation against Ubuntu 18, and Ubuntu 20 and while I can get things to work fine on a VM,  the builds fail in my TravisCI tests at the libicu stage. 

**Why:** 

In a stock Ubuntu Installation there is a softlink from a .so to the specific version installed. E.g. 

```
lrwxrwxrwx 1 root root      16 Mar 17  2020 /usr/lib/x86_64-linux-gnu/libicuio.so -> libicuio.so.66.1
```

In QCGPostLinkCommon.pri the specific version is hard coded to 
```
QT_LIB_LIST += \
        libicudata.so.56 \
        ...
```

This matches the files that are delivered in the TravisCI dependencies tarfile download from S3 ( https://s3-us-west-2.amazonaws.com/qgroundcontrol/dependencies/Qt5.12.6-gcc_64-min.tar.bz2 , specified in the .travis.yml file ) 

However the lack of a soft link in the tar file differs in how one typically sees the libraries delivered by packages. 

**Solution:**
I'm guessing the fix for this and a bunch of other open issues I've seen regarding various debian versions (e.g. Ubuntu 16, 18, 20) and compiling qgroundcontrol is to resolve this difference in soft link existence in libicu libraries between packages and the dependency file. 

There are a couple of potential ways to do this

I think the way that will result in the minimal # of changes is as follows: 

1.   For the dependencies tarfile, make softlinks in the directory as `ln -s libicudata.so.56 libicudata.so`  (and so-on for the other libicu libraries)
2.   Create a tar file that preserves links
3.   Re-deploy this tar file to the S3 AWS site
4.   Change QCGPostLinkCommon.pri to point to .so instead of .so.VERSION

I can't create a pull request that you can pull into this repo because I don't have access to the S3 location. 

Another way would be to specify in the .travis.yml file the installation of the dependencies directly (e.g. apt install, git pull) and not have them in the .tar file. This seems like the most amount of work.  

Another way would be to put into the .travis.yml file something like this command (not tested)
```
cd /tmp/path/to/lib/
for FOO in $(ls $.so.56); do BAR=${FOO/.56/}; ln -s $FOO $BAR; done
```
but that would break with each change in the dependencies file. 

If you do one of these, I'd be glad to test it if you want.

AJRepo, 2020-10-21:
Oops. I may have jumped the gun on my last comment.  

I've re-downloaded the tar file and now in looking at the directory Qt5.12-gcc_64/5.12.6/gcc_64/lib/  I'm seeing the soft links like `libicudata.so -> libicudata.so.56.1` , so I guess I'll withdraw my comment about changing the dependencies file/directory.

AJRepo, 2020-10-21:
Looking more closely at this: 

with this setting:

```
QT_LIB_LIST += \
        libicudata.so \
        libucui18n.so \
        libicuuc.so
```

the "linux installer" compiles ok (minus the error that it can't upload) but the TravisCI "linux debug" can't run the deploy without this error

if [[ "${SPEC}" = "linux-g++-64" && "${CONFIG}" = "debug" ]]; then mkdir -p ~/.config/QtProject/ && cp ${TRAVIS_BUILD_DIR}/test/qtlogging.ini ~/.config/QtProject/ && export QT_FATAL_WARNINGS=1 && ./staging/qgroundcontrol-start.sh --unittest; fi
**/tmp/shadow_build_dir/staging/QGroundControl: error while loading shared libraries: libicuuc.so.56: cannot open shared object file: No such file or directory**

The file  libicuuc.so.56 I found in the dpendencies tar file ( `Qt5.12.6-gcc_64-min.tar.bz2` )  

Using ldd on  the files in that tar file I found  some libicu files in the tar file are statically linked and others dynamically linked
and ldd reports "*not found** errors .   Perhaps this is why you have to specify libicudata.so.56 specifically instead of libicudata.so to get the build to work? 

```
cd /tmp/
wget https://s3-us-west-2.amazonaws.com/qgroundcontrol/dependencies/Qt5.12.6-gcc_64-min.tar.bz2
tar -jxf Qt5.12.6-gcc_64-min.tar.bz2
cd /tmp/Qt5.12-gcc_64/5.12.6/gcc_64/lib

for FOO in $(ls libicu*); do echo $FOO; ldd $FOO; done

libicudata.so
        statically linked
libicudata.so.56
        statically linked
libicudata.so.56.1
        statically linked
libicui18n.so
        linux-vdso.so.1 (0x00007ffde7b1d000)
        libicuuc.so.56 => not found
        libicudata.so.56 => not found
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f9fd2d24000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f9fd2d1e000)
        libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f9fd2b3d000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f9fd29ee000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f9fd29d1000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f9fd27df000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f9fd3218000)
libicui18n.so.56
        linux-vdso.so.1 (0x00007ffd87feb000)
        libicuuc.so.56 => not found
        libicudata.so.56 => not found
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fa1e7ac9000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fa1e7ac3000)
        libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fa1e78e2000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fa1e7793000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fa1e7776000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fa1e7584000)
        /lib64/ld-linux-x86-64.so.2 (0x00007fa1e7fbd000)
libicui18n.so.56.1
        linux-vdso.so.1 (0x00007fff72dde000)
        libicuuc.so.56 => not found
        libicudata.so.56 => not found
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fc4b8739000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fc4b8733000)
        libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fc4b8552000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fc4b8403000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fc4b83e6000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fc4b81f4000)
        /lib64/ld-linux-x86-64.so.2 (0x00007fc4b8c2d000)
libicuio.so
        linux-vdso.so.1 (0x00007ffd8ceef000)
[... 267 lines cut ...]
        libicui18n.so.56 => not found
        libicuuc.so.56 => not found
        libicudata.so.56 => not found
libicutest.so.56.1
        libicui18n.so.56 => not found
        libicuuc.so.56 => not found
        libicudata.so.56 => not found
libicutu.so
        libicui18n.so.56 => not found
        libicuuc.so.56 => not found
        libicudata.so.56 => not found
libicutu.so.56
        libicui18n.so.56 => not found
        libicuuc.so.56 => not found
        libicudata.so.56 => not found
libicutu.so.56.1
        libicudata.so.56 => not found
libicuuc.so
        libicudata.so.56 => not found
libicuuc.so.56
        libicudata.so.56 => not found
libicuuc.so.56.1
        not a dynamic executable
        not a dynamic executable
```

AJRepo, 2020-11-17:
If I understand the core issue here then I think I fixed this issue with this commit:  https://github.com/mavlink/qgroundcontrol/compare/master...AJRepo:libicu_fix

Changes: 
1) Call .so instead of a specific so.56 (see QGCPostLinkCommon.pri ) . This also solves the build problems on other Ubuntu versions (e.g. Ubuntu 20.04) 

2) Update ldconfig to include downloaded library. On TravisCI build (see .travis.yml file ) the files are extracted to /tmp/.... I don't know if that needs to be elsewhere for the full released version. 

With 1 and 2 above the TravisCI build works. 

Let me know if this is ok for creating a pull request

DonLakeFlyer, 2020-11-19:
> 1. Call .so instead of a specific so.56

So this change to .56 was for some specific reason. Can you go through the git history and figure out why/when this was changed. Simply changing it back doesn't seem to make sense since it was changed away from that for some reason. Not sure why that reason is somehow no longer valid.

FlavioTonelli, 2020-11-19:
If it can help this is the last discussion about the same topic: https://github.com/mavlink/qgroundcontrol/issues/7382

AJRepo, 2020-11-19:
Looks like it's related to this git update: https://github.com/mavlink/qgroundcontrol/commit/10bf1605b5f6d52b5859bb4f160daf044a46e921

I followed your link to #7382 and looked in the last discussions for someone running ldconfig or ldd .  I didn't see that so would guess that not running ldconfig is the root cause, which you can get around by specifying .56 so the linker doesn't have to follow the softlinks (and getting 'not found' errors). 

I see someone said 

> "Looks like the bundled Qt libraries link explicitly to libicui18n.so.56, libicuuc.so.56 and libicudata.so.56. The AppImage used to contain these libraries with their version suffixes, but now it contains these libraries without the suffixes."  

I don't see where the "bundled Qt libraries" are. I see this download in .travis.yml : https://s3-us-west-2.amazonaws.com/qgroundcontrol/dependencies/Qt5.12.6-gcc_64-min.tar.bz2 and if that's the one, then that's what I've been testing against in TravisCI. 

My guess is that 
  (1) there was a requirement to use the newer Qt libraries (.56 or greater) but older Ubuntu versions (e.g 16, 18) didn't provide a new enough package version 
and 
  (2) when the separate .56 libraries were downloaded, they weren't added to the ldconfig path and thus if you only specified .so instead of .56 you get a "not found" error. 

You can replicate this test with ldd and ldconfig by typing "ldd filename" before and after adding the directory where the files are to /etc/ld.so.conf.d/somename.conf and running ldconfig. 

I seem to recall that if I *didn't* use ldconfig in the .travis.yml file then the only way I could get it to compile was specifying .56. A successful build in specifying .so  happened only if you also update the library path with ldconfig.

DonLakeFlyer, 2020-11-19:
So there are two ways QGC is released to users:
* A tarball you expand and then run using the qgroundcontrol-start.sh
* An AppImage built by https://github.com/mavlink/qgroundcontrol/blob/master/deploy/create_linux_appimage.sh in travis.yml

If the tarball version is the only thing causing problems I would drop support for it. I've been wanting to do that for a while now. That requires changing .travis.yml to run unit tests using the AppImage version of Qt. Also something I've wanted to do since it validates the AppImage build to some extent as well in CI. Currently is AppImage build doesn't work it just a silent failure caught by the first poor schmuck who happens to install a new daily. All of which I haven't had time to getting around to. If anyone here could help with moving to that it would be great.

DonLakeFlyer, 2020-11-21:
FYI: I'm trying to switch QGC unit testing to use the AppImage. I'm having problems with Qt 5.15 testing which I think will fix this as well.

AJRepo, 2020-11-21:
> I'm having problems with Qt 5.15 testing

In which branch are you are doing the testing? I'd like to see how you installing Qt 5.15. In the 5.12.6 Qt install (above) the tarball downloaded from AWS/S3 failed unless you ran ldconfig to include the extracted files. 

Both TravisCI builds ( "Linux Debug" and a "Linux Installer")  seemed to compile fine for me as .so except that the "Linux Installer"  gave an error at the end that "The request signature we calculated does not match the signature you provided. Check your key and signing method."  ( https://travis-ci.org/github/AJRepo/qgroundcontrol/jobs/744217662#L3988 ).

DonLakeFlyer, 2020-11-21:
It's in here. The .travis.yml will point you to the 5.15 qt tarball

DonLakeFlyer, 2020-11-21:
> "The request signature we calculated does not match the signature you provided. Check your key and signing method."

That's because you are building from your own repo on travis. It's complaining about signatures for upload to S3 I believe which you can't do from your repo because you don't have the keys.

DonLakeFlyer, 2020-11-21:
> In the 5.12.6 Qt install (above) the tarball downloaded from AWS/S3 failed unless you ran ldconfig to include the extracted files.

Interesting. So after the tarball is extracted should .travis scripts do an ldconfig in there? I know next to nothing about Linux so don't really understand what that does.

AJRepo, 2020-11-22:
>  So after the tarball is extracted should .travis scripts do an ldconfig in there? I know next to nothing about Linux so ...

While my experience with linux admin is high, my experience on the compiler end is much more elementary and I'm just beginning to look at qgroundcontrol here so hopefully I'm not going to mangle this explanation too badly ... 

The .so file on linux is a "shared object" file or the equivalent of a "dynamic linked library."  So if you want to compile a program on linux and specify the generic .so you need to update the path where the dynamic libraries are installed. That's what `ldconfig` is for. 

When I looked at the libraries in the Qt tar files with `ldd` it showed the libraries with the .so as being dynamically linked and that's why it failed compiling with .so instead of .so.XX even though the .so file IS in the AWS-S3 tarfile. 

An easy way is to update that path is to add a config file to /etc/ld.so.conf.d/  and then run ldconfig. 

See the diff on the .travis.yml  here: https://github.com/AJRepo/qgroundcontrol/commit/d423dd1bf38e3e61132c1c33841b609ce0facc7e#diff-6ac3f79fc25d95cd1e3d51da53a4b21b939437392578a35ae8cd6d5366ca5485

Or to be specific,  see lines 109-110 in .travis.yml in that commit. 
```
        echo "/tmp/Qt5.12-gcc_64/5.12.6/gcc_64/lib" | sudo tee -a /etc/ld.so.conf.d/qt512.conf &&
        sudo ldconfig &&
```

Note I hardcoded the version, so really it's `echo "/tmp/Qt${QTVERSION}-gcc_64/${QTVERSION}/gcc_64/lib"`

The caveat is that if you compiled using dynamic libraries AND the executable expects these external dynamic libraries (e.g. you compile a smaller executable with libraries not compile- in statically) then those dynamic libraries have to also be on the system (and in the ldconfig path) where the program is deployed. 

You say 

> I'm trying to switch QGC unit testing to use the AppImage.

I've not really looked much at your end deployment methods, but I'm going to **guess** that if your goal is package deployment then if you can specify an apt repo instead of the tarfile download from S3-AWS, then that might make the final deployment go smoother because (1) you can specify that library version (or higher) as a requirement and then the package manager will handle making sure the shared libraries are at the correct version and (2) the package manager handles lddconfig stuff for you. 

Ubuntu 18 (which is what TravisCI has) natively has qtbase-dev = 5.9.5 which I think isn't a high enough version and I didn't see any higher versions released as packages for Ubuntu 18.  ( https://wiki.qt.io/Install_Qt_5_on_Ubuntu#Installation_Guide_.28Ubuntu_package.29 ) 

For compiling qgroundcontrol on Ubuntu 20.04 I just used the native package  qtbase5-dev = 5.12.8 which worked just fine, 

As an aside: I saw this note on the QT.io wiki

 > This is valid until 5.14.2 due to policy change, distributing Open Source Qt linux package is discontinue from 5.15.0. You need to build your own package starting from 5.15.0. 

and I'm not quite sure what they mean by distributing Open Source Qt linux package is discontinue from 5.15.0.

DonLakeFlyer, 2020-11-23:
> This is valid until 5.14.2 due to policy change, distributing Open Source Qt linux package is discontinue from 5.15.0. You need to build your own package starting from 5.15.0.

I can't seem to fine this. Do you have a link?

AJRepo, 2020-11-23:
> I can't seem to fine this. Do you have a link?

On the wiki:  Under the section "Installation Guide (Qt download page)"

https://wiki.qt.io/Install_Qt_5_on_Ubuntu#Installation_Guide_.28Ubuntu_package.29

AJRepo, 2020-11-23:
I see there are is a QT library snap install ( https://snapcraft.io/qt551 )  but I don't know if those snaps are "official" or not.

acxz, 2021-02-18:
@LorenzMeier  can u close this issue? Resolved with https://github.com/mavlink/qgroundcontrol/commit/cc95825594fc99e7537198003cab4a0dd1172bcb

acxz, 2021-02-20:
@LorenzMeier another friendly ping, I think you may have just forgot to look at this issue a couple days ago.

closed 2021-02-20 by LorenzMeier