mavlink/qgroundcontrol #8334: MAVCMD retransmit an infinite number of time

FlavioTonelli, 2020-02-14:
In Vehicle::_handleCommandAck() the function _sendNextQueuedMavCommand() is called every time an ack is received.

In the case,GCS continuously receive a stream of ack that are not related to the MAVCMD QGC is trying to transmit then the retransmission logic break. GCS start retransmit the current CMD an infinity number of time, blocking all the other MAVCMD in queue.

DonLakeFlyer, 2020-02-16:
@FlavioTonelli If you have a repro, can you verify this fix?

DonLakeFlyer, 2020-02-16:
Reopen due to discussion on pull.

FlavioTonelli, 2020-02-16:
It seems it is working fine now. Adding or not the target id field is not preventing this issue to be closed. I still suggest to implement that to be more compliant to the MAVLink specification.

DonLakeFlyer, 2020-02-17:
> I still suggest to implement that to be more compliant to the MAVLink specification.

That's still marked as WIP which generally means don't use.

closed 2020-02-19 by DonLakeFlyer