mavlink/qgroundcontrol #8343: 3.5.6 Driver Power State Failure

Jai-GAY, 2020-02-16:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Please tell us what should happen as detailed as possible.

## Current Behavior
Please describe what happens instead of the expected behavior.

Driver power State Failure when shutting down windows 10 64 bits. no hardware needed to test.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. install version 3.5.6 64 bits
2. launch QGC
3. exit QGC
4. REPEAT step 2 and 3 few times
5. open task manager, notice QGC exist in background
6. shutdown window 10 64 bits
7. blue screen Driver Power State Failure

## System Information
When posting bug reports, include the following information
- Operating System: [e.g. iOS 12.1, Windows 8, macOS Mojave 10.14.1, Ubuntu 14] windows 10 64 bits
- QGC Version: [e.g. 3.4.4] 3.5.6 64 bits
- QGC build: [e.g., daily, stable, self-built from source, etc...] stable
- Flight Controller: [e.g., Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.], n.a.
- Autopilot (with version): [e.g., PX4 1.8.1, Ardupilot Copter 3.6.1] n.a.

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

[image]

Jai-GAY, 2020-02-21:
[tested 3.5.5](https://github.com/mavlink/qgroundcontrol/releases/tag/v3.5.5), same problem.

Jai-GAY, 2020-02-27:
[v4.0.0 still crash my 64bit windows 10](https://github.com/mavlink/qgroundcontrol/releases/tag/v4.0.0) when doing reboot or shutdown

[image]

[image]

DonLakeFlyer, 2020-03-01:
QGC 4.0 is only supported on 64 bit systems. The WIndows Protected Your PC messages is due to the fact that the QGC installer is not signed. You need to select "Run Anyway".

DonLakeFlyer, 2020-03-01:
First thing is to make sure you are testing with 4.0 release. Then you notice that you exit QGC and see that it is still in the Task Manager. Can you then start QGC again? EVen though there is one already running

Jai-GAY, 2020-03-06:
Clearly I mentioned shut down the system.

DonLakeFlyer, 2020-03-06:
> Clearly I mentioned shut down the system.

I understand that. I am trying to debug the problem. I need to know what happens when you try to start QGC again when task manager says it is already running.

DonLakeFlyer, 2020-03-11:
> I need to know what happens when you try to start QGC again when task manager says it is already running.

@Jai-GAY Any possibility of trying this out?

Jai-GAY, 2020-03-16:
[4.0.3 is the worst of 4.x release](https://github.com/mavlink/qgroundcontrol/releases/tag/v4.0.3)

[this video shows I can't even shut down the computer.](https://drive.google.com/file/d/1wJ6xY8JJBJntNwhm1ILc0vp5w7tEcLCr/view?usp=sharing) I just did a fresh install, restart the computer, next start QGC 4.0.3, and then the entire system hangs, can't even reboot, sign out or restart the system.

[image]

DonLakeFlyer, 2020-03-16:
A hunch would be video driver problems. YOu can try starting QGC with one of the other GPU startup options:
[image]

This will bypass your video driver.

Jai-GAY, 2020-03-31:
tested [4.0.5 QGroundControl (GPU Compatibility Mode) and QGroundControl (GPU Safe Mode)](https://github.com/mavlink/qgroundcontrol/releases/tag/v4.0.5), nothing different. still, crash my window 10 during shut down.

[image]

DonLakeFlyer, 2020-03-31:
> I am trying to debug the problem. I need to know what happens when you try to start QGC again when task manager says it is already running.

Also still waiting to hear back on this.

Jai-GAY, 2020-04-06:
> > I am trying to debug the problem. I need to know what happens when you try to start QGC again when task manager says it is already running.
> 
> Also still waiting to hear back on this.

I do not understand why I must do this. As I had repeatedly able to reproduce the problem using identical steps/sequences. I am reporting about the first launch after installation and shut down the computer.

DonLakeFlyer, 2020-04-06:
> I do not understand why I must do this.

Because it will help me try to understand what is going wrong! 

From your report:
2. launch QGC
3. exit QGC
4. REPEAT step 2 and 3 a few times

I need to understand whether step 3 leaves a running QGC according to task manager, and yet you can still launch QGC again. That indicates a different level of shutdown problem.

Jai-GAY, 2020-06-06:
I [test version 4.0.8](https://github.com/mavlink/qgroundcontrol/releases/tag/v4.0.8) it seems to work one time, meaning I can start QGC, exit properly, restart QGC, exit properly, Each time there is no QGC running in the background, exit properly. restart PC fast and no problem. BUT, while I return to remove 4.0.8 and install 4.0.1, it crashes the entire system completely. then I return to test 4.0.8 again and the problem resurfaced. The Driver Power State Failure seems to be surrounded by the wireless power settings and if QGC did not exit properly (meaning it is still alive in the background), my guess. I got 4.0.8 working fine once is because I changed to use Balanced (recommended) power options instead of Panasonic Power management (i suspect the cause of it).

[image]

Jai-GAY, 2020-06-09:
I cannot produce this problem on another 64bits Windows OS (Home) machine.

[image]

Jai-GAY, 2021-01-29:
[v4.1.1 windows 64bits](https://github.com/mavlink/qgroundcontrol/releases/tag/v4.1.1), I still have the DRIVER POWER STATE FAILURE issue when I shut down the computer.

step 1. install QGC
step 2. reboot computer
step 3. start v4.1.1
step 4. pause for few mins
step 5. close QGC
step 6. shutdown computer
step 7. system hangs, Driver Power State Failure