mavlink/qgroundcontrol #8347: Request: eliminate need for [Apply Preset] button in Surveys

Antiheavy, 2020-02-16:
The [Apply Preset] button is awkward and inconsistent with other QGC drop down selectors. 
[image]

Most other drop down selectors in QGC operate by changing immediately to the item selected.  This is the most intuitive method and most drop down selectors in other applications generally work this way.

I think most people expect drop down selectors to do 2 things:
a) work as selector buttons
b) indicate current state

The current Presets drop down doesn't do either of the above.  The fact that it may not show the correct state is the most concerning to us (our users will be upset when the vehicle flies the wrong altitude and overlap, even though the Preset words were shown right in their face on screen).

Suggestions:
[image]

1) Preset drop down could initially show "_select a preset_" in italics or somehow to indicate nothing has been selected. (there are probably other methods that can work here too)
2) allow the drop-down menu to apply the settings when picked.  This way the preset drop down also shows current state.
3) if someone tabs away and then back, it should revert back to "_select a preset_" since you don't know what the user has changed in the mean-time.  Then if a drop down is selected it can be trusted to be showing the correct state again.
4) remove the [Apply Preset] button
5) suggestion: a simpler [New Preset] button should be enough to get the point across, especially since the saving/naming dialog provides extra context:
[image]

This should apply to the Survey Wizard Preset selector the same way.

Antiheavy, 2020-02-20:
Examples of other drop-down selectors in QGC that work the way I'm requesting here:

Camera selector:
[image]

Camera settings in Mission Start pane:
[image]

[image]

Flight Mode selector:
[image]

Many of the General Settings options:
[image]

Antiheavy, 2020-02-24:
FYI @dakejahl what do you think?

dakejahl, 2020-02-24:
Yup I think this makes a lot of sense, it should be straight forward to change.

DonLakeFlyer, 2020-02-24:
> it should be straight forward to change.

I thought there was some backend architecural problem with this. Which is why I did it that way. I'd have to go digging for it again for remember. It's not as simple as changing a few things.

DonLakeFlyer, 2020-02-25:
I think there is still some backend problem which I need to look into further. 

But the main driver is user model. I originally had it working the other way and had to rewrite the whole thing because I could not get it to work. When you select "Apply Preset" that is all it does, "Apply". It just changes a pile of values to the values defined in the preset. After you've done that, there is no longer an understanding of where those values came from. The Preset is not saved with the Plan. Making the Preset "apply" by just selecting it from the Combo gives the incorrect impression that the preset is sticky/saved. For example, I applied a preset. Save to file. Reload the file, the preset is no longer selected. Or I select a preset. I then change a value somewhere and forgot about it. I then go to look at Presets page and my original preset is still selected. So I think I'm fine, everything is set according to the preset when it is not. That is the fundamental difference between the concept of "Apply". After it's done the concept of where those values came from is lost. I can see how that difference will be lost on most people, but without the Apply I think it's even worse from a confusion standpoint.

DonLakeFlyer, 2020-02-25:
The other problem is that you can't manage your presets without trashing your current Plan. In order to delete a preset you need to select it. Not as a big of a problem but still there.

In reality the whole thing needs a rework which is why I marked this as 4.1

Antiheavy, 2020-02-25:
> For example, I applied a preset. Save to file. Reload the file, the preset is no longer selected. Or I select a preset. I then change a value somewhere and forgot about it. I then go to look at [...]

Yep, all of this is totally right.  My hope was that by showing "_Select a preset_" every time you switch into the Preset tab we are not making any assumptions at all about anything currently on-screen matching a preset configuration.
[image]

Would "_Apply a preset_" work better?

Antiheavy, 2020-04-03:
> The other problem is that you can't manage your presets without trashing your current Plan. In order to delete a preset you need to select it. Not as a big of a problem but still there.

In this scenario maybe the [Delete Preset] button is grayed out unless a preset is actually selected in the drop-down.