mavlink/qgroundcontrol #8357: Request/ Inquiry: Multiple locations single vehicle

simba86, 2020-02-19:
### TL;DR
I wish to put two location indicators on the map for a single vehicle.
### Details
We are using a forked version of QGC to control autonomous vehicles developed in our lab (I'm a student), with our own autopilots and hardware.
Each of the vehicles always reports two different positions to the QGC. One is based on GPS and the other on an INS (inertial navigation system) estimate.
This is done in order to develop our own INS.

So far in order to have two locations, we always used two system ID's, (two vehicles) to distinguish the locations.

Is there way to have to locations on map for a single vehicle? I can use separate Mavlink component ID's to distinguish them in post processing, but not in . the map?
If I would wish to implement this, do you have suggestions regarding how t go about this?
I don't want to be to invasive, and avoid deviating too much from the main development path.

DonLakeFlyer, 2020-02-19:
So have the Vehicle object report a new position property. And then change the map code to display something at that location as well.

closed 2020-02-19 by DonLakeFlyer

simba86, 2020-02-20:
> So have the Vehicle object report a new position property. And then change the map code to display something at that location as well.

Thanks