mavlink/qgroundcontrol #8365: UDP broadcasts not working on Android

lucasdemarchi, 2020-02-19:
When running ardupilot with the ":bcast" flag, it sends the mavlink data to the broadcast address and when the first "client" responds it locks down to the address that responded. This works fine when running QGC on my Linux machine, but it's not working on Android.

I think this is related to this QT issue https://bugreports.qt.io/browse/QTBUG-73138 and the workaround is to acquire  WifiManager.MulticastLock (it's a global lock, so any app acquiring this will make other apps to work).

From @DonLakeFlyer:

    Hmm, maybe. I know it used to work. But may have broken with change to 5.11. Enter an issue and I'll try to reproduce.

DonLakeFlyer, 2020-02-21:
> When running ardupilot with the ":bcast" flag, it sends the mavlink data to the broadcast address and when the first "client" responds it locks down to the address that responded.

What's the SITL command line to do taht?

DonLakeFlyer, 2020-03-12:
> What's the SITL command line to do taht?

@lucasdemarchi Can you help here?

lucasdemarchi, 2020-03-12:
Sorry for the delay, I use it with Linux boards, not sitl... But I think sitl can do that too. I will check here later today and update.

DonLakeFlyer, 2020-03-12:
I think people are seeing this with Solos and Google Pixel devices. Want to verify that it is android version specific problem. I don't have a newer Android 8 device so I can't test that well.

lucasdemarchi, 2020-03-12:
Yes, I have a pixel too. But AFAIR I had the same behavior on a Amazon Fire tablet.

DonLakeFlyer, 2020-03-12:
> Amazon Fire tablet.

What version of Android is on there. May be Android 8 and above only.

DonLakeFlyer, 2020-03-12:
Also do you get no connection at all? Seems like some folks are getting Vehicle->QGC traffic but no QGC->Vehicle traffix.

PR #9199 (2020-11-24): Take MulticastLock on Android to ensure broadcast packets can be received
Workaround for QTBUG-73138
fixes #8365
closed 2020-11-24 by DonLakeFlyer