mavlink/qgroundcontrol #8372: UI scaling on macOS

andrewvoznytsa, 2020-02-20:
If I switch macOS display settings like on attached screenshot 
[image]
 then QGC does not render its UI properly - see this attachment 
[image]
. Playing with variables described here https://doc.qt.io/qt-5/highdpi.html did not help.

macOS - Catalina
QGC - master

DonLakeFlyer, 2020-02-20:
Is this scaling a new Catalina thing? I'm guessing it's a Qt problem which is dealt with in newer version.

andrewvoznytsa, 2020-02-20:
No, scaling is old feature.

dogmaphobic, 2020-02-21:
Something is peculiar about your setup. The (System Settings) UI looks different. It appears it is not able to collect all the information about the monitor. Is that running on a VM? I tested changing to all possible scaling options and it all works (Qt does get notified and repaints itself).

[image]
[image]
[image]

andrewvoznytsa, 2020-02-21:
Yes, it was VM. I'll try on real HW in a few days.

DonLakeFlyer, 2020-11-06:
Reopen if happens in non-VM

closed 2020-11-06 by DonLakeFlyer