mavlink/qgroundcontrol #8374: Draft: QGroundControl Stable 4.0 available

DonLakeFlyer, 2020-02-20:
Here is a very high level and non-exhaustive list of new features added to QGroundControl in version 4.0. Not to mention the large number of bug fixes in this release.

* Settings
  * Language: Allow selection of language
  * ArduPilot
    * Mavlink: Configurable stream rate settings
* Setup
  * Joystick
    * New joystick setup ui
    * Ability to configure held button to single or repeated action
  * ArduPilot
    * Follow Me support for ArduCopter
    * Motor Test
* Plan
  * Create Plan from template with wizard like progression for completing full Plan.
  * Survey: Save commonly used settings as a Preset
  * Polygon editing
    * New editing tools ui
    * Support for tracing a polygon from map locations
* Fly
  * Click to ROI support
  * Added support for connecting to ADSB SBS server. Adds support for ADSB data from USB SDR Dongle running 'dump1090 --net' for example.
  * Ability to turn on Heading to home, COG and Next Waypoint heading indicators in Compass.
  * Video
    * Add support for UDP RTP h.265 video streams
* Analyze
  * New Mavlink Inspector which includes charting support
* General
  * Released Windows build are now 64 bit only
  * Log Replay: Ability to specify replay speed

Docs and Downloads here: http://qgroundcontrol.com

qter188, 2020-02-21:
Wow, amazing!

jsharpe, 2020-02-21:
The macOS build needs to be notarized as per instructions here: https://developer.apple.com/documentation/xcode/notarizing_macos_software_before_distribution as it doesn't work on Catalina even with the instructions given.

qter188, 2020-02-21:
Is it released? Why the the file is v3.5.6 still.

DonLakeFlyer, 2020-02-21:
@dogmaphobic Any ideas on Catalina then? I can't test on my system since I've already turned it all off.

dogmaphobic, 2020-02-21:
Just this morning I downloaded something (unsigned). It is as I described. You just need to insist. From Finder, right click on the .app bundle and select open. It will chastise you about it and the only option is _Cancel_. Do it again, this time you have an _Open_ button available. You only need to do this once.

jsharpe, 2020-02-21:
Unfortunately this only works if your gatekeeper settings are configured to allow this.
We have our devices configured according to NCSC guidelines: https://www.ncsc.gov.uk/collection/mobile-device-guidance/platform-guides/macos which adds a policy to turn off the ability to run untrusted apps.

Is the issue that you don't have access to an apple developer certificate to do the signing?

jsharpe, 2020-02-21:
> Just this morning I downloaded something (unsigned). It is as I described. You just need to insist. From Finder, right click on the .app bundle and select open. It will chastise you about it and the [...]

Following this process with the MDM policy set to turn off the ability to run untrusted apps results in the following dialog instead of one that gives you an open button:
https://slack-files.com/T0V5F863B-FU3J63WDN-f98202bcca

dogmaphobic, 2020-02-21:
> Is the issue that you don't have access to an apple developer certificate to do the signing?

It's more than just that. The only way right now to produce a signed binary is if I manually build it using my personal developer's account. The existing (automated) build system is not setup to do that. This is all volunteer work. We will gladly accept help in setting up a build system that produces signed binaries.

DonLakeFlyer, 2020-02-21:
> Is it released?

Not yet

DonLakeFlyer, 2020-02-25:
Note: I pulled ArduPilot follow me support from stable. It just wasn't ready.

closed 2020-02-26 by DonLakeFlyer