mavlink/qgroundcontrol #8395: Corridor Wizard: [Trace] gets stuck in Corridor Scan too

Antiheavy, 2020-02-22:
Similar to what was happening in Survey (https://github.com/mavlink/qgroundcontrol/issues/8344), Trace in Corridor Scan keeps active even when you are "Done" with the Corridor Scan and move on to Landing Pattern or other items.

The fix for Survey is effective https://github.com/mavlink/qgroundcontrol/pull/8373 as it grays out the [Done With Polygon] until you click [Done Tracing].  However,  it would be less annoying and easier to use if [Done With Polygon] and [Done Polyline] also performed the [Done Tracing] function as well.  This way the user has the option to click either [Done Tracing] or [Done With Polygon]/[Done Polyline] and the effect is the same, it stops the tracing and they move on to whichever their next thing is.

DonLakeFlyer, 2020-02-22:
> Trace in Corridor Scan keeps active even when you are "Done" with the Corridor Scan

Ah, crap I should have none that!

DonLakeFlyer, 2020-02-23:
> However, it would be less annoying and easier to use if [Done With Polygon] and [Done With Polyline] also performed the [Done Tracing] function as well.

Can you enter a separate feature request for that. I'm only fixing bug now. Don't want to break as much as I fix. I'll change that after stable goes out.

closed 2020-02-23 by DonLakeFlyer