mavlink/qgroundcontrol #8401: C2220 error in QGCTileCacheWorker.cpp, many files produce C4996 and C7567 errors.

zk317, 2020-02-24:
<C2220 error in QGCTileCacheWorker.cpp. Version 4.0 upgrade error?>

Hey, I'm having fun watching your work.

I wanted to try it too, but when I built it, I ran into problems.
(I didn't fix anything)

I get a C2220 error in QGCTileCacheWorker.cpp

In addition, many files produce C4996 and C7567 errors.

I installed it as described in the link below.

https://dev.qgroundcontrol.com/en/getting_started/

I installed Visual Basic 2017 version Desktop development with C ++ and qt 5.12.5 MSVC 2017 64 64bit.

git clone <https://github.com/mavlink/qgroundcontrol.git --recursive>

Download the source with the command.

I tried to reinstall it several times, but I get a comprehensive error such as C2220 error.

I seems to be something wrong with the version.

If you look for these errors on Google, and you don't see the same case, it might be because of a version update?

....

I found it in gitter
<I read that "QGC is currently built against Qt 5.12.6 (all OS) .Also Windows has moved to VS 2017 and 64 bit only.Docs updates may be a bit behind since I haven't had time.>

So I deleted and reinstalled Qt5.12.5.
same.

I rebuilt the window.
same.

It sure seems like a compiler problem.

I tried to reinstall the compiler with version 15.9 of 2015, so I tried version 15.0.
same.

I also checked Qt's kit several times.

[image]

What is the problem?
Are you going too fast?

DonLakeFlyer, 2020-02-24:
I don't think you have the right compilers set up:
[image]

zk317, 2020-02-25:
Thanks for the answer
If I solve this problem, I'll work hard to update the development guide sheet!

But right now the problem is not solved ...

I didn't get to version 15.9, but I did it again after seeing the answer you gave me.
I wanted to install the compiler language in English
It didn't work out ... (see screenshot)

[image]

I think now
- Qt5.12.6 installation from the web (I proceeded with a 3.6GB offline installation from the archive)
- I'm going to install Visual Studio 2017 15.9 other than Community.
- I'm thinking about moving to Linux if that doesn't work.

zk317, 2020-02-25:
Hi guys!

As mentioned above, I have reinstalled qt, the compiler, and Windows with various versions and roots.

Eventually it failed with the same error.

But I did not give up.

We have installed Ubuntu Linux.

[image]

amazing!
New updates every day!

Unfortunately I can't build a development environment on Windows.
I'm glad you're doing well on Linux.

By the way, I don't know how to distribute Linux.
I know exactly Windows, but I'm new to Linux.

So .... Why can't only hybrids work in the mapbox?
I confirmed that it works fine on Windows.

DonLakeFlyer, 2020-02-25:
Wait I didn't realize C2220 turns warnings into errors. You can do that with current QGC source. Did you turn that on yourself?

zk317, 2020-02-26:
I didn't understand exactly what you mean.
What do you want me to do?
Are you going to replace the error with a warning and run it?
I'm ready to come to Windows anytime!

But on Linux, it crashes because of an irregular crashed error.
Master is the latest version
I don't know what's wrong with the error.
Are there any logs left except qt messages?

zk317, 2020-02-26:
If you run it with .sh you get an error code,
I'll add this to the issue category separately.

And "Wait I didn't realize C2220 turns warnings into errors.You can do that with current QGC source. Did you turn that on yourself?" If you explain what this means, I'll help you as much as I can.

I'm always thankful

DonLakeFlyer, 2020-02-27:
Sorry I forgot that the Windows build compiles by default with WarningsAsErrors on so it compiles with the /WX flag. Which will fail the build with a C2220 error if there is a compiler warning. It looks like the compiler you are using is newer than the one we are using hence it has new warning checks in it which are now failing which didn't before.

I think the reason may be because of this: "Visual Basic 2017 version Desktop development with C ++ and qt 5.12.5 MSVC 2017 64 64bit." The VB version may install something newer than if you just use the regular C++ install as outlined in the docs.

DonLakeFlyer, 2020-02-27:
Did some more digging looks like the compiler was updated in newer VS 2017 versions. So these warnings are new in that version. I'll get them fixed up.

DonLakeFlyer, 2020-02-27:
I upgraded VS and got a new compiler. What I get from the standard C++ install is still a little older then what you have. And when I recompile I sitll don't get those errrors. Hmm

zk317, 2020-03-02:
I didn't monitor this post because I knew it couldn't be fixed.
I erased all the files on Windows.
I will test it in Windows environment and leave a reply as soon as possible.

qter188, 2020-03-07:
You should remove the -WX option of CXXFLAGS from the makefile, which on the output directory of Qt Creator. @zk317

zk317, 2020-04-08:
>You should remove the -WX option of CXXFLAGS from the makefile, which on the output directory of Qt Creator.

-------------------------------------

thank you. Well solved.

I have confirmed that it works fine on Windows.

Thank you all.

closed 2020-04-08 by zk317

dhruv-mehta99, 2020-04-29:
i am a noob in Qground control i m also getting this c2220 error so finally how should i sovle it?

zk317, 2020-05-16:
In my case, it was solved by setting WX option of CXXFLAGS.
If you install the master version now, it will work well without taking the above measures.