mavlink/qgroundcontrol #8405: Tuning page PID/Advanced: Improvement suggestions

MaEtUgR, 2020-02-25:
Follow up to https://github.com/mavlink/qgroundcontrol/issues/6438

- What problem does it solve?
People want easy setup and since tuning is part of the initial setup of a new vehicle we should have a UI to simplify the process for newcomers but also developers in general. I realized there is already a quite nice tuning page and I'm just collecting suggestions to improve it while testing.

- What flight stacks must it work with? (All, PX4, ArduPilot)
I'm testing with PX4

- Relevant vehicle types? (All, multirotor, fixed-wing, VTOL, submarine, etc.)
Multirotor

- What does it do?
The plots are awsome and the plus minus buttons to adjust all significant gains on the side is **so useful!**

The following are just suggestions I noted during my first tests:
1. "Advanced" checkbox in the top right corner is not visible/clear enough. Honestly, I initially overlooked it.
1. When tuning roll and pitch automatic mode switching 
Start: Altitude mode
Stop: Position mode
Allows to not having to deal with holding altitude manually while giving step inputs and allows to bring back the drone in position after a tuning session.
1. `MC_ROLLRATE_MAX` and `MC_PITCHRATE_MAX` should definitely be included in that tuning since while tuning you see the limits being hit very well in the plot and it should be the goal of the tuning to not expect to hit those limits.
1. I don't see how `MC_ROLLRATE_FF` and  `MC_PITCHRATE_FF` are useful to a multicopter tuning. It's for helicopter use only see https://github.com/PX4/Firmware/pull/9289#issuecomment-380464077. We need to make that more clear also from the parameter description and documentation (FYI @hamishwillee).
1. It would be nice to have a sync button for roll and pitch. More often than not the tuning is very similar for roll and pitch with just slight adjustments depending on mass distribution on the vehicle. My workflow would be to tune roll first then take over the gains to pitch and adjust whatever necessary.
1. The absolute yaw plot has some yaw wrapping problems, they could in principle already be on the firmware site:
[image]
1. I realized while tuning yaw that sometimes parameters set in the fields get saved or resaved when pressing the start button. Is it that the "new" gain set in the fields only apply in between pressing start and stop? It's a bit strange when you had to force a parameter but it keeps on showing the dialog to foce it again every time you press start.
1. I managed to crash QGC while on the tuning page, first I couldn't switch to any other page anymore and then it segfaulted with
```
free(): double free detected in tcache 2
/tmp/.mount_AuteriEAgQ4T/AppRun: line 10: 20227 Aborted                 (core dumped) "${HERE}/AuterionGS" "$@"
```
This made me think about the mode should probably be switched back whenever data link is lost which is not that easy from the UI.

DonLakeFlyer, 2020-02-25:
Thanks for feedback @MaEtUgR. Let me start by saying I myself have no idea really how to correctly tune a vehicle, nor have I ever done it. My first cut at this was to try to create something that even I could use and understand. With hopes that someone (like you!) could help to make it real. Just explaining so you understand you are likely to run into things which make no sense at all.

I'll try to get back to this soon. Fighting with Stable now.

MaEtUgR, 2020-03-02:
@DonLakeFlyer Totally, I assumed that and the page with the live plots is really useful even as it is and I'll definitely use it to bring up the next vehicle. I was just trying to provide feedback after talking to @dagar and knowing how much of a difference such UI will have on people's tuning.

> I'll try to get back to this soon. Fighting with Stable now.

Cool, feel free to ask me any time for follow up. The yaw wrapping issue could be on the autopilot side since the ground station should not need to adjust the setpoints shown.

closed 2026-10-04 by github-actions[bot]