mavlink/qgroundcontrol #8407: where the "Fly" and "Plan" labels removed intentionally?

Antiheavy, 2020-02-25:
I've noticed in recent Daily Builds (maybe going back a while) the "Fly" and "Plan" labels at the top of the left-hand toolbars are no longer there.  Where these removed intentionally?
[image]

The documentation still talks about Fly and Plan views, which if there are no longer any labels in the software could be confusing:
https://docs.qgroundcontrol.com/en/FlyView/FlyView.html
https://docs.qgroundcontrol.com/en/PlanView/PlanView.html
FYI @hamishwillee

hamishwillee, 2020-02-25:
Probably an improvement. Either way we should do a full scan through the docs and update quite a lot of the images. I'm hoping other people step up for some of that.

DonLakeFlyer, 2020-02-25:
Hmm, let me look at when those got pulled out

DonLakeFlyer, 2020-02-25:
No, these should not have been pulled out

Antiheavy, 2020-02-26:
Looks like "Fly" and "Plan" are back in the latest Daily.  I agree, it is better to have the words there. Closing this issue.
[image]

closed 2020-02-26 by Antiheavy