mavlink/qgroundcontrol #8415: Loading KML and SHP files moves map to 0,0 Africa

Antiheavy, 2020-02-25:
In recent Daily builds whenever we load a KML or SHP file for a Survey the map view flies to lat,lon 0,0 off the coast of Africa.  The KML and SHP "seems" to load correctly at the proper location.  This happens on both KML and SHP files which are known to work fine with older versions of QGC. 

[image]

DonLakeFlyer, 2020-02-25:
Can you link to one that breaks. Then ones I"m testing are working.

Antiheavy, 2020-02-26:
[Joe 160.zip](https://github.com/mavlink/qgroundcontrol/files/4252942/Joe.160.zip)

[Farm Demo.zip](https://github.com/mavlink/qgroundcontrol/files/4252945/Farm.Demo.zip)

Attached "Joe160" SHP and "Farm Demo" KML.  Both of these work normally with older QGC version.  

To reproduce (I've done this on 3 different Windows machines now):
1) Plan View
2) File -> Survey
3) Click [Done Adjusting] for Takeoff
3) Click ]Load KML/SHP...]
4) Browse to file, click Open
5) Observe map pans to 0,0 off coast of Africa
6) Observe the Survey is correctly created according to the KML/SHP

closed 2020-02-26 by DonLakeFlyer