mavlink/qgroundcontrol #8422: Only mapbox's Hybrid "Failed to fetch tile error"

zk317, 2020-02-26:
Hey, I'm back

This time is new troble for mapbox api.

I compile last version fo V4.0.0

and then I tryed Hybrid of Mapbox.

but Hybird not open.

Map window is all White.

[image]

however I tryed open qgroundcontrol-start.sh
so, cmd window  printed  error messege

That error is....

QGeoTileRequestManager: Failed to fetch tile (437,201,9) 5 times, giving up. Last error message was: 'Error transferring https://api.mapbox.com/v4/mapbox.hybrid/9/437/201.jpg80?access_token=pk.eyJ1IoimsTciLhIjoiYs1Z0dWMdgrhWWhejetjyHWBDyGZkZnnd9.pGcTOreRTftSmb_uKoM - server replied: Not Found'

Another Mapbox map style is woking good.

1. How can I fix this bug?

2. Why map is blurred little bit? (my map not sharpness, especily text) How can I perpect sharp

(pls click Picture look detail)

[image]
[image]

dogmaphobic, 2020-03-03:
> Error transferring https://api.mapbox.com/v4/mapbox.hybrid/9/437/201.jpg80?access_token=pk.eyJ1IoimsTciLhIjoiYs1Z0dWMdgrhWWhejetjyHWBDyGZkZnnd9.pGcTOreRTftSmb_uKoM - server replied: Not Found'

That URL gives me an "invalid token" error. Is that token real or you just made something up to paste it here? If it is real, the issue seems to be with how you have your account setup.

zk317, 2020-04-08:
I put random values in the token.
For security reasons

Also, the problem was solved by adding a personal editable api in Mapbox Studio.

Thanks for the reply

closed 2020-04-08 by zk317