mavlink/qgroundcontrol #8425: QGC Track Replay on macOS 10.15.2 window jumps between positions (3.5.5 and 3.5.6)

dlwalter, 2020-02-26:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Windows stays in position during replay

## Current Behavior
During tlog replay the windows jumps between two positions making screen capture difficult

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Open QGC on macOS
2. Select File -> Replay Flight Data
3. Select 'Replay Flight Data' on bottom of screen
4. Select a tlog
5. Play
6. Observe window

## System Information
When posting bug reports, include the following information
- Operating System: macOS 10.15.2
- QGC Version: 3.5.5 and 3.5.6
- QGC build: downloaded from release page
- Flight Controller: Pixhawk Cube
- Autopilot (with version): PX4 v1.9.2

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots

[image]

DonLakeFlyer, 2020-02-26:
Fixed in daily builds

closed 2020-02-26 by DonLakeFlyer

dlwalter, 2020-02-26:
This may be a dumb question - but has the "replay flight log" selection moved in the daily build?  I just installed it and don't have File or Analyze on the menu bar.  I can't figure out how to replay a track on the daily build.

[image]

DonLakeFlyer, 2020-02-26:
Yup. Settings/General/Fly