mavlink/qgroundcontrol #8430: Extra photos taken during turning in px4

moreba1, 2020-02-27:
Hi

In turn around there is extra photos taken , the issue is in planing survey mission for px4 code .

[image]

DonLakeFlyer, 2020-02-27:
Can you attach the plan file you are using?

moreba1, 2020-02-28:
> 
> 
> Can you attach the plan file you are using?

yes , i attached plan . i think camera trigger distance must be separately after start waypoint not in camera section .

[plan.zip](https://github.com/mavlink/qgroundcontrol/files/4265500/plan.zip)

DonLakeFlyer, 2020-02-28:
I looked at your plan and the mission generated has the correct camera start and stop command and the beginning and end of each transect. So this would be some sort of either firmware or camera problem. You'll need to talk with firmware folks.

Antiheavy, 2020-03-01:
@moreba1 This is a know issue with PX4. @LorenzMeier and @DonLakeFlyer did a bunch of work on it a while back, but it seems to have been tricky to solve and maybe was dropped. https://github.com/PX4/Firmware/pull/11878

LorenzMeier, 2020-03-01:
@DonLakeFlyer Have you implemented MAV_CMD_CONDITION_GATE consistently by now? If not, that part should still be open on the QGC side, as you otherwise have no clean definition of the transect.

DonLakeFlyer, 2020-03-01:
> This is a known issue with PX4.

This is not related to MAV_CMD_CONDITION_GATE. The current problem with turning off images in turnaround is that the images stop too early not too late. This is because the vehicle hits the acceptance radius which triggers the state machine to move to the next step which is to stop triggering. This in turn can happen early enough that you miss photos at the edge of the survey area.

What is shown above is the opposite problem. The photos stop to late and in one case there is even a strange gap after transect exit. You can see from how the flight path begins to curve towards the turnaround that the mission state machine has moved past the survey edge waypoint which in turn means it also has processed the camera stop. Yet camera camera did not stop.

moreba1, 2020-03-01:
> What is shown above is the opposite problem. The photos stop to late and in one case there is even a strange gap after transect exit. You can see from how the flight path begins to curve towards the [...]

my problem is not  "images stop too early" , my problem is images start too early that cause we have images with 30 degree angles .

[image]

DonLakeFlyer, 2020-03-01:
Ooh, crap. Sorry not paying enough attention when I looked at your plan file. I thought the flight path was going the other direction. So this is the MAV_CMD_CONDITION_GATE implementation problem. Let me look at that again.

moreba1, 2020-05-21:
does it done ?

DonLakeFlyer, 2020-05-24:
Yes, but I'm not quite sure where the firmware support stands.

moreba1, 2020-05-25:
> 
> 
> Yes, but I'm not quite sure where the firmware support stands.

@LorenzMeier