mavlink/qgroundcontrol #8431: QGC 4.0 Flashing Ardupilot Chibios Developer Build

dkemxr, 2020-02-27:
When attempting to Flash a Kakute F7 AIO with a Chibios Developer build the only Board Type that shows in the down down is for Slyviper.

DonLakeFlyer, 2020-02-27:
Can you go through the same steps with FirmwareUpgradeLog turned on and attach the console log output?

https://docs.qgroundcontrol.com/en/SettingsView/console_logging.html

dkemxr, 2020-02-27:
Here you go Don.

 
 

https://docs.qgroundcontrol.com/en/SettingsView/console_logging.html

dkemxr, 2020-02-27:
Here you go Don

 
 

https://docs.qgroundcontrol.com/en/SettingsView/console_logging.html

  
[D] at C:\projects\qgroundcontrol\src\QGCApplication.cc:261 - "Settings location "C:/Users/dkemxr/AppData/Roaming/QGroundControl.org/QGroundControl.ini" Is writable?: true"
[D] at C:\projects\qgroundcontrol\src\QGCLoggingCategory.cc:99 - "Filter rules "*Log.debug=false\n""
[D] at C:\projects\qgroundcontrol\src\Vehicle\MAVLinkLogManager.cc:360 - "MAVLink logs directory: "C:/Users/dkemxr/Documents/QGroundControl/Logs""
[D] at C:\projects\qgroundcontrol\src\QGCApplication.cc:361 - "System reported locale: QLocale(English, Default, United States) "en_US""
[D] at C:\projects\qgroundcontrol\src\QGCApplication.cc:435 - "Loading localization for "en_US""
[D] at C:\projects\qgroundcontrol\src\QtLocationPlugin\QGCMapEngine.cpp:164 - "Map Cache in: "C:/Users/dkemxr/AppData/Local/cache/QGCMapCache300" / "qgcMapCache.db""
[D] at C:\projects\qgroundcontrol\src\QGCApplication.cc:809 - ""v4.0.0""
[D] at C:\projects\qgroundcontrol\src\QGCLoggingCategory.cc:99 - "Filter rules "*Log.debug=false\nFirmwareUpgradeLog.debug=true\n""
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\PX4FirmwareUpgradeThread.cc:368 - "PX4FirmwareUpgradeThreadController::findBoard"
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:827 - "Found px4 stable version "v1.10.1 Stable Release""
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:832 - "Found px4 beta version "v1.10.0 First Release Candidate""
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:872 - "_ardupilotManifestDownloadFinished "http://firmware.ardupilot.org/manifest.json" "C:/Users/dkemxr/AppData/Local/Temp/manifest.json""
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:197 - ""STMicroelectronics." "STMicroelectronics Virtual COM Port""
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:212 - "0 "ArduPilot ChibiOS""
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\PX4FirmwareUpgradeThread.cc:204 - "_findBootloader "\\\\.\\COM22""
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\PX4FirmwareUpgradeThread.cc:249 - "Found bootloader"
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:727 - "_buildAPMFirmwareNames"
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:748 - "Fallback match: "KakuteF7 - 4.0.2" 1155 22336 123 "https://firmware.ardupilot.org/Copter/stable/KakuteF7/arducopter.apj" FirmwareUpgradeController::CopterFirmware"
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:748 - "Fallback match: "Pixhawk1-1M - 4.0.2" 1155 22336 123 "https://firmware.ardupilot.org/Copter/stable/Pixhawk1-1M/arducopter.apj" FirmwareUpgradeController::CopterFirmware"
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:748 - "Fallback match: "Omnibus F7 V2 - 4.0.2" 1155 22336 123 "https://firmware.ardupilot.org/Copter/stable/OMNIBUSF7V2/arducopter.apj" FirmwareUpgradeController::CopterFirmware"
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:748 - "Fallback match: "KakuteF4 - 4.0.2" 1155 22336 123 "https://firmware.ardupilot.org/Copter/stable/KakuteF4/arducopter.apj" FirmwareUpgradeController::CopterFirmware"
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:748 - "Fallback match: "Pixhawk1 - 4.0.2" 1155 22336 123 "https://firmware.ardupilot.org/Copter/stable/Pixhawk1/arducopter.apj" FirmwareUpgradeController::CopterFirmware"
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:748 - "Fallback match: "CubeSolo - 4.0.2" 1155 22336 123 "https://firmware.ardupilot.org/Copter/stable/CubeSolo/arducopter.apj" FirmwareUpgradeController::CopterFirmware"
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:748 - "Fallback match: "Matek F405 Wing - 4.0.2" 1155 22336 123 "https://firmware.ardupilot.org/Copter/stable/MatekF405-Wing/arducopter.apj" FirmwareUpgradeController::CopterFirmware"
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:748 - "Fallback match: "skyviper-journey - 4.0.2" 1155 22336 123 "https://firmware.ardupilot.org/Copter/stable/skyviper-journey/arducopter.apj" FirmwareUpgradeController::CopterFirmware"
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:748 - "Fallback match: "fmuv2 - 4.0.2" 1155 22336 123 "https://firmware.ardupilot.org/Copter/stable/fmuv2/arducopter.apj" FirmwareUpgradeController::CopterFirmware"
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:748 - "Fallback match: "Pixhawk 3 Pro - 4.0.2" 1155 22336 123 "https://firmware.ardupilot.org/Copter/stable/DrotekP3Pro/arducopter.apj" FirmwareUpgradeController::CopterFirmware"
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:748 - "Fallback match: "fmuv5 - 4.0.2" 1155 22336 123 "https://firmware.ardupilot.org/Copter/stable/fmuv5/arducopter.apj" FirmwareUpgradeController::CopterFirmware"
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:748 - "Fallback match: "SpeedyBee F4 - 4.0.2" 1155 22336 123 "https://firmware.ardupilot.org/Copter/stable/speedybeef4/arducopter.apj" FirmwareUpgradeController::CopterFirmware"
[... cut ...]
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:727 - "_buildAPMFirmwareNames"
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:748 - "Fallback match: "skyviper-v2450 - 4.0.3" 1155 22336 123 "https://firmware.ardupilot.org/Copter/latest/skyviper-v2450/arducopter.apj" FirmwareUpgradeController::CopterFirmware"
[D] at qrc:/qml/FirmwareUpgrade.qml:387 - "model [skyviper-v2450 - 4.0.3]"
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\PX4FirmwareUpgradeThread.cc:374 - "PX4FirmwareUpgradeThreadController::cancel"

DonLakeFlyer, 2020-02-27:
What OS are you using? I need to get you a new build with some more logging in it.

dkemxr, 2020-02-27:
Windows 10

DonLakeFlyer, 2020-02-27:
A new build should be available here: https://qgroundcontrol.s3-us-west-2.amazonaws.com/builds/Stable_V4.0/QGroundControl-installer.exe in about 10 minutes. Use that and do the same thing again. You should see a change in the output of this line:
```
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:727 - "_buildAPMFirmwareNames"
```
with extra info.

If you don't then you still ahve the old build.

dkemxr, 2020-02-27:
I uninstalled 4.0 (basic Windows uninstall) and installed the new build. Log attached after trying firmware update.

 
 

  
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:727 - "_buildAPMFirmwareNames "STMicroelectronics Virtual COM Port" 1155 22336"
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:727 - "_buildAPMFirmwareNames "STMicroelectronics Virtual COM Port" 1155 22336"

DonLakeFlyer, 2020-02-27:
So far it's pointing to missing items in the manifest file. But still working to verify.

DonLakeFlyer, 2020-02-27:
Here is what I can see from the log:
* The board port description is coming across as "STMicroelectronics Virtual COM Port". I would have expected it to be "KakuteF7-BL". Which is what I believe the latest windows drivers and the ArduPilot bootloader should return. But I may be wrong there.
* The VID/PID for the board is 0x0483/0x5740 which correctly identified it as a random ArduPilot chibios board.
* Since the bootloader string doesn't match is falls back to VID/PID matching in the manifest.
* The next set of log information shows the list generated for a stable build which matches these parameters. Here you can see a big list of possible firmwares which is correct. Since QGC has limited information as to what the board is. Generic VID/PID matching will cause that.
* Then I can see you switch to Developer builds. Which kicks off rebuilding the firmware list.
* Now I can see that only the skyviper shows up. Which as far as I can tell means that there is a bug in the firmware manifest file.

So two problems:
* Why isn't the board reported as "KakuteF7-BL" by OS
* Even without that it should still work, although the filtering will be crap. Which indicates a manifest bug.

@tridge Can you take a look at what I've written up here?

tridge, 2020-03-10:
>     * The board port description is coming across as "STMicroelectronics Virtual COM Port". I would have expected it to be "KakuteF7-BL".

those are two quite different things. The "STMicroelectronics Virtual COM Port" is a driver ID, not the product string ID. The two don't come from the same place.

> Since the bootloader string doesn't match is falls back to VID/PID matching in the manifest.

VID/PID matching does not usefully narrow down the list of boards for most boards. You really need to get the board_id as that is what ultimately identifies the hardware and thus the list of valid firmwares. 

>    Now I can see that only the skyviper shows up. Which as far as I can tell means that there is a bug in the firmware manifest file.

I don't understand that conclusion. As far as I can see the manifest is correct and I don't see how you end up with a skyviper-v2450 which is board_id 9 (it is an fmuv3 hw board) for a kakutef7.

DonLakeFlyer, 2020-03-10:
> The "STMicroelectronics Virtual COM Port" is a driver ID, not the product string ID. The two don't come from the same place.

@tridge Given the fact that QGC uses Qt as the base cross platform library. I'm distances from the actual OS port values. I have access to what is called a description and a manufacturer. When I plug in a CubeBlack for example with the ArduPilot bootloader I get this:
* description - CubeBlack-BL, then CubeBlack
* manufacturer - Hex/ProfiCNC

The description field is what I used to build the firmware lookups in the manifest. It is what I used to use to build the firmware URLs directly from the board info with the old code prior to switching to use the manifest.

Yet with this testing against a KakuteF7 it is reporting as VID/PID using the generic ArduPilot VID/PID values of 0x0483/0x5740 which I assume means it is running the ArduPilot bootloader. If it is running the ArduPilot bootloader then when aren't I seeing KakuteF7-BL (which would then in turn pick the right firmware) in description field of the port like I see for all the other boards I've tested with.

DonLakeFlyer, 2020-03-10:
@dkemxr Just to make sure there isn't some bug in FirmwareUpgrade port reporting. Can you turn on just LinkManagerVerboseLog, connect the KakuteF7 and attach the console output here. That will give me full info from the port.

DonLakeFlyer, 2020-03-10:
> VID/PID matching does not usefully narrow down the list of boards for most boards. You really need to get the board_id as that is what ultimately identifies the hardware and thus the list of valid firmwares.

@tridge I thought the whole point of the ArduPilot bootloader was to not have to do board id matching any more? Hence the VID/PID onyl lookup is just a fallback when the bootloader match fails.

tridge, 2020-03-10:
> When I plug in a CubeBlack for example with the ArduPilot bootloader I get this:
> 
>     * description - CubeBlack-BL, then CubeBlack
> 
>     * manufacturer - Hex/ProfiCNC

right, and those are the strings we generate in ArduPilot. The string "STMicroelectronics Virtual COM Port" has never been used by ArduPilot. That is a driver string for a windows driver and has never been present in any ArduPilot version. Perhaps the Qt libraries are conflating the two?

> The description field is what I used to build the firmware lookups in the manifest. It is what I used to use to build the firmware URLs directly from the board info with the old code prior to switching to use the manifest

that is all good, but that "STMicroelectronics" string is not a description string that comes from the running firmware.

> If it is running the ArduPilot bootloader then when aren't I seeing KakuteF7-BL (which would then in turn pick the right firmware) in description field of the port like I see for all the other boards I've tested with.

The ArduPilot bootloader must indeed be running, and it will be giving a sane description and product string. What must be happening is that some layer between QGC and windows is screwing that up and giving you a completely different string, the driver name string, which is completely useless for board identification.

I still don't understand how QGC came up with "skyviper" though. There is nothing in the manifest linking any of the above information with "skyviper"

dkemxr, 2020-03-10:
Here you go Don.

[D] at C:\projects\qgroundcontrol\src\QGCApplication.cc:261 - "Settings location "C:/Users/dkemxr/AppData/Roaming/QGroundControl.org/QGroundControl.ini" Is writable?: true"
[D] at C:\projects\qgroundcontrol\src\QGCLoggingCategory.cc:99 - "Filter rules "*Log.debug=false\n""
[E] at :0 - "serialnmea: No known GPS device found. Specify the COM port via QT_NMEA_SERIAL_PORT."
[D] at C:\projects\qgroundcontrol\src\Vehicle\MAVLinkLogManager.cc:360 - "MAVLink logs directory: "C:/Users/dkemxr/Documents/QGroundControl/Logs""
[D] at C:\projects\qgroundcontrol\src\QGCApplication.cc:361 - "System reported locale: QLocale(English, Default, United States) "en_US""
[D] at C:\projects\qgroundcontrol\src\QGCApplication.cc:435 - "Loading localization for "en_US""
[D] at C:\projects\qgroundcontrol\src\QtLocationPlugin\QGCMapEngine.cpp:164 - "Map Cache in: "C:/Users/dkemxr/AppData/Local/cache/QGCMapCache300" / "qgcMapCache.db""
[D] at C:\projects\qgroundcontrol\src\comm\MAVLinkProtocol.cc:224 - "Switching outbound to mavlink 2.0 due to incoming mavlink 2.0 packet: 0x7ff795b70ed0 2 2"
[D] at C:\projects\qgroundcontrol\src\comm\MAVLinkProtocol.cc:467 - "Temp log "C:/Users/dkemxr/AppData/Local/Temp/FlightData174094.mavlink""
[E] at :0 - "libpng warning: iCCP: known incorrect sRGB profile"
[D] at C:\projects\qgroundcontrol\src\Vehicle\Vehicle.cc:2671 - "_parametersReady true"
[D] at C:\projects\qgroundcontrol\src\QGCLoggingCategory.cc:99 - "Filter rules "*Log.debug=false\nLinkManagerVerboseLog.debug=true\n""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:520 - "-----------------------------------------------------"
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:521 - "portName:           "COM8""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:522 - "systemLocation:     "\\\\.\\COM8""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:523 - "description:        "u-blox Virtual COM Port""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:524 - "manufacturer:       "u-blox AG""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:525 - "serialNumber:       """
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:526 - "vendorIdentifier:   0"
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:527 - "productIdentifier:  0"
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:520 - "-----------------------------------------------------"
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:521 - "portName:           "COM1""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:522 - "systemLocation:     "\\\\.\\COM1""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:523 - "description:        "Communications Port""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:524 - "manufacturer:       "(Standard port types)""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:525 - "serialNumber:       """
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:526 - "vendorIdentifier:   0"
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:527 - "productIdentifier:  0"
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:520 - "-----------------------------------------------------"
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:521 - "portName:           "COM22""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:522 - "systemLocation:     "\\\\.\\COM22""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:523 - "description:        "STMicroelectronics Virtual COM Port""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:524 - "manufacturer:       "STMicroelectronics.""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:525 - "serialNumber:       "2C002B001751343439333839""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:526 - "vendorIdentifier:   1155"
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:527 - "productIdentifier:  22336"
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:567 - "Skipping existing autoconnect "\\\\.\\COM22""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:520 - "-----------------------------------------------------"
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:521 - "portName:           "COM8""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:522 - "systemLocation:     "\\\\.\\COM8""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:523 - "description:        "u-blox Virtual COM Port""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:524 - "manufacturer:       "u-blox AG""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:525 - "serialNumber:       """
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:526 - "vendorIdentifier:   0"
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:527 - "productIdentifier:  0"
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:520 - "-----------------------------------------------------"
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:521 - "portName:           "COM1""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:522 - "systemLocation:     "\\\\.\\COM1""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:523 - "description:        "Communications Port""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:524 - "manufacturer:       "(Standard port types)""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:525 - "serialNumber:       """
[... cut ...]
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:524 - "manufacturer:       "STMicroelectronics.""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:525 - "serialNumber:       "2C002B001751343439333839""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:526 - "vendorIdentifier:   1155"
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:527 - "productIdentifier:  22336"
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:567 - "Skipping existing autoconnect "\\\\.\\COM22""

DonLakeFlyer, 2020-03-10:
> What must be happening is that some layer between QGC and windows is screwing that up and giving you a completely different string, the driver name string, which is completely useless for board identification.

That would be the first time I've seen anything like that. Which is why I wondered whether there may be a bug in the Windows drivers from Michael. Not sure if that could screw something up?

@tridge Do you someone in Kakute land that could send me one of these for testing? I don't have one.

> I still don't understand how QGC came up with "skyviper" though. There is nothing in the manifest linking any of the above information with "skyviper"

I'm still digging through this again. I haven't gotten that far yet.

DonLakeFlyer, 2020-03-10:
Ok so this:
```
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:523 - "description: "STMicroelectronics Virtual COM Port""
[D] at C:\projects\qgroundcontrol\src\comm\LinkManager.cc:524 - "manufacturer: "STMicroelectronics.""
```

Shows that the firmware upgrade code is using the information it was given. It's not some bug in there where things get screwed up as the strings are passed around.

tridge, 2020-03-10:
>  I thought the whole point of the ArduPilot bootloader was to not have to do board id matching any more? Hence the VID/PID onyl lookup is just a fallback when the bootloader match fails.

The correct process to use is to always at some point in the process go via the board_id. That is the hardware id - it is the only thing you can use to work out what possible firmwares can be offered for a board.
The question then is how to find the board_id. There are a few methods depending on what process the user goes through in QGC to load a firmware:
 - if you can get the board into the bootloader then you can ask the bootloader for it's board id. This approach requires that you get the board into the bootloader before you narrow down what firmwares can be used. It is the most reliable method, but may not suit the UI flow you have in QGC
 - you can look at the product and description strings. If they match ones that are in the manifest then you need to look at the board_id for those matching entries, then use that board_id as explained below
Once you have the board_id you should use just that one number to give a list of firmwares that have that board_id. For some board_id values (eg. 9 for fmuv3) this will be quite a long list. It would be nice if the default element or first element in the sorted list corresponded to the firmware that they currently have loaded.

tridge, 2020-03-10:
> Shows that the firmware upgrade code is using the information it was given

those strings didn't come from ArduPilot though. I suspect something screwy in the windows driver layer that is hiding the strings the device is providing and replacing them with something else.
This is where the method of starting the bootloader protocol *before* you select which firmware to download is such a big win. It bypasses the whole driver issue and gets you the real board_id, which gets you the list of firmwares that are hardware compatible with this specific board.

DonLakeFlyer, 2020-03-10:
> This is where the method of starting the bootloader protocol _before_ you select which firmware to download is such a big win.

That is how QGC works. You wouldn't get to the firmware selection dialog unless it already had the full bootloader info and was stopped waiting in bootloader mode.

tridge, 2020-03-10:
> That is how QGC works. You wouldn't get to the firmware selection dialog unless it already had the full bootloader info and was stopped waiting in bootloader mode.

that is great! Then you should do this:
 - request the board_id from the bootloader using PROTO_DEVICE_BOARD_ID of PROTO_GET_DEVICE
 - use that board_id to match all entries in the manifest that have that board_id. This is the list of hw compatible firmwares
 - if you have a string description from USB that matches one of those matching firmwares then list this as the first (or default) option in the selection list of available firmwares

DonLakeFlyer, 2020-03-10:
I already have that information and used to use it and then I screwed up the code when I switched to the manifest.

DonLakeFlyer, 2020-03-10:
> * use that board_id to match all entries in the manifest that have that board_id. This is the list of hw compatible firmwares

If I have a bootloader string match, why provide the user with anything other than that? Why would they want to use one of the other firmware choices?

DonLakeFlyer, 2020-03-10:
> You can't shortcut and avoid the board_id step as the firmware the user has loaded may not be the right firmware. For example, due to previous screwups in firmware loaders users commonly are running "fmuv5" where they should be running "Pixhawk4" or are running "fmuv3" where they should be running "CubeBlack".

I think that answers my above question. Is that right?

DonLakeFlyer, 2020-03-10:
@tridge Is there a way to distinguish fmuv2 and fmuv3 firmwares in the manifest? These both report as fmuv2 with only the flash size distinguishing the different. How to I prevent a user from flashing an fmuv2 build onto an fmuv3 board. And the opposite as well. Also if I know this is an fmuv3 which I do, how can I narrow the list available to only fmuv3 builds?

DonLakeFlyer, 2020-03-10:
> Is there a way to distinguish fmuv2 and fmuv3 firmwares in the manifest?

The best I can see now is to filter on fmuv2 in the platform name.

DonLakeFlyer, 2020-03-10:
I also don't understand how just doing a board id match leads to good results for filtering. For example here is a CubeBlack:
```
2020-03-10 12:05:09.618763-0700 QGroundControl[54663:6917644] [FirmwareUpgradeLog] Board id match: "CubePurple - 4.0.3" "CubeBlack-BL" ("CubePurple-BL", "PX4 BL FMU v2.x") "https://firmware.ardupilot.org/Copter/stable/CubePurple/arducopter.apj" FirmwareUpgradeController::CopterFirmware
2020-03-10 12:05:09.618794-0700 QGroundControl[54663:6917644] [FirmwareUpgradeLog] Board id match: "Pixhawk1-1M - 4.0.3" "CubeBlack-BL" ("Pixhawk1-1M-BL", "PX4 BL FMU v2.x") "https://firmware.ardupilot.org/Copter/stable/Pixhawk1-1M/arducopter.apj" FirmwareUpgradeController::CopterFirmware
2020-03-10 12:05:09.618829-0700 QGroundControl[54663:6917644] [FirmwareUpgradeLog] Board id match: "Pixhawk1 - 4.0.3" "CubeBlack-BL" ("Pixhawk1-BL", "PX4 BL FMU v2.x") "https://firmware.ardupilot.org/Copter/stable/Pixhawk1/arducopter.apj" FirmwareUpgradeController::CopterFirmware
2020-03-10 12:05:09.618858-0700 QGroundControl[54663:6917644] [FirmwareUpgradeLog] Board id match: "CubeSolo - 4.0.3" "CubeBlack-BL" ("CubeSolo-BL", "PX4 BL FMU v2.x") "https://firmware.ardupilot.org/Copter/stable/CubeSolo/arducopter.apj" FirmwareUpgradeController::CopterFirmware
2020-03-10 12:05:09.618882-0700 QGroundControl[54663:6917644] [FirmwareUpgradeLog] Board id match: "CubeBlack - 4.0.3" "CubeBlack-BL" ("CubeBlack-BL", "PX4 BL FMU v2.x") "https://firmware.ardupilot.org/Copter/stable/CubeBlack/arducopter.apj" FirmwareUpgradeController::CopterFirmware
2020-03-10 12:05:09.618907-0700 QGroundControl[54663:6917644] [FirmwareUpgradeLog] Bootloader best match: "CubeBlack - 4.0.3" "CubeBlack-BL" ("CubeBlack-BL", "PX4 BL FMU v2.x") "https://firmware.ardupilot.org/Copter/stable/CubeBlack/arducopter.apj" FirmwareUpgradeController::CopterFirmware
2020-03-10 12:05:09.618946-0700 QGroundControl[54663:6917644] [FirmwareUpgradeLog] Board id match: "skyviper-journey - 4.0.3" "CubeBlack-BL" ("skyviper-journey-BL", "PX4 BL FMU v2.x") "https://firmware.ardupilot.org/Copter/stable/skyviper-journey/arducopter.apj" FirmwareUpgradeController::CopterFirmware
2020-03-10 12:05:09.618974-0700 QGroundControl[54663:6917644] [FirmwareUpgradeLog] Skipping fmuv2 manifest entry for fmuv3 board: "fmuv2 - 4.0.3" "CubeBlack-BL" ("fmuv2-BL", "PX4 BL FMU v2.x") "https://firmware.ardupilot.org/Copter/stable/fmuv2/arducopter.apj" FirmwareUpgradeController::CopterFirmware
2020-03-10 12:05:09.618997-0700 QGroundControl[54663:6917644] [FirmwareUpgradeLog] Board id match: "skyviper-v2450 - 4.0.3" "CubeBlack-BL" ("skyviper-v2450-BL", "PX4 BL FMU v2.x") "https://firmware.ardupilot.org/Copter/stable/skyviper-v2450/arducopter.apj" FirmwareUpgradeController::CopterFirmware
2020-03-10 12:05:09.619024-0700 QGroundControl[54663:6917644] [FirmwareUpgradeLog] Board id match: "CubeGreen Solo - 4.0.3" "CubeBlack-BL" ("CubeGreen-solo-BL", "PX4 BL FMU v2.x") "https://firmware.ardupilot.org/Copter/stable/CubeGreen-solo/arducopter.apj" FirmwareUpgradeController::CopterFirmware
2020-03-10 12:05:09.619051-0700 QGroundControl[54663:6917644] [FirmwareUpgradeLog] Board id match: "mRo X2.1 - 4.0.3" "CubeBlack-BL" ("mRoX21-BL", "PX4 BL FMU v2.x") "https://firmware.ardupilot.org/Copter/stable/mRoX21/arducopter.apj" FirmwareUpgradeController::CopterFirmware
2020-03-10 12:05:09.619077-0700 QGroundControl[54663:6917644] [FirmwareUpgradeLog] Board id match: "fmuv3 - 4.0.3" "CubeBlack-BL" ("fmuv3-BL", "PX4 BL FMU v2.x") "https://firmware.ardupilot.org/Copter/stable/fmuv3/arducopter.apj" FirmwareUpgradeController::CopterFirmware
```

Not all the random non-CubeBlack thing listed here. Throwing that in front of the user is going to be really confusing.

tridge, 2020-03-10:
> I think that answers my above question. Is that right?

yes, it is quite common that a user has the wrong firmware installed on a board, especially fmuv3. This particularly happens as many boards ship with the older generic px4 bootloader, and firmware loading tools (eg. MissionPlanner) tended to select fmuv3 as the firmware to load.

tridge, 2020-03-10:
> Note all the random non-CubeBlack things listed here

yes. That is the worst one of course, and it really stems from a poor decision many years ago to make an "fmuv3" architecture that covers a huge range of very different hardware. Many of these should have had separate board_id values.
You could narrow down the list somewhat using the image_size tag, at least removing firmwares that won't fit. 
Otherwise the only idea I have is to have some special cases for the "generic" firmwares like fmuv3, fmuv4 and fmuv5. If the string matches those then show the full list. Otherwise if the string matches a non-generic name then show just the matching one but have a button to expand the list to include all matching boards. If it would help I could add a "generic" tag to the manifest to allow automation of this choice.

DonLakeFlyer, 2020-03-11:
@dkemxr Can you try this new (and as yet unreleased) stable: https://qgroundcontrol.s3-us-west-2.amazonaws.com/builds/Stable_V4.0/QGroundControl-installer.exe

dkemxr, 2020-03-11:
Well, that seemed to work Don. Firmware update log as follows:

[D] at C:\projects\qgroundcontrol\src\QGCApplication.cc:261 - "Settings location "C:/Users/dkemxr/AppData/Roaming/QGroundControl.org/QGroundControl.ini" Is writable?: true"
[D] at C:\projects\qgroundcontrol\src\QGCLoggingCategory.cc:99 - "Filter rules "*Log.debug=false\n""
[D] at C:\projects\qgroundcontrol\src\Vehicle\MAVLinkLogManager.cc:360 - "MAVLink logs directory: "C:/Users/dkemxr/Documents/QGroundControl/Logs""
[D] at C:\projects\qgroundcontrol\src\QGCApplication.cc:361 - "System reported locale: QLocale(English, Default, United States) "en_US""
[D] at C:\projects\qgroundcontrol\src\QGCApplication.cc:435 - "Loading localization for "en_US""
[D] at C:\projects\qgroundcontrol\src\QtLocationPlugin\QGCMapEngine.cpp:164 - "Map Cache in: "C:/Users/dkemxr/AppData/Local/cache/QGCMapCache300" / "qgcMapCache.db""
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:878 - "_ardupilotManifestDownloadFinished "http://firmware.ardupilot.org/manifest.json" "C:/Users/dkemxr/AppData/Local/Temp/manifest.json""
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:197 - ""ArduPilot Project" "ArduPilot""
[D] at C:\projects\qgroundcontrol\src\QGCLoggingCategory.cc:99 - "Filter rules "*Log.debug=false\nFirmwareUpgradeLog.debug=true\n""
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\PX4FirmwareUpgradeThread.cc:368 - "PX4FirmwareUpgradeThreadController::findBoard"
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:833 - "Found px4 stable version "v1.10.1 Stable Release""
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:838 - "Found px4 beta version "v1.10.0 First Release Candidate""
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:878 - "_ardupilotManifestDownloadFinished "http://firmware.ardupilot.org/manifest.json" "C:/Users/dkemxr/AppData/Local/Temp/manifest.json""
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:197 - ""ArduPilot Project" "ArduPilot""
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:212 - "0 "ArduPilot ChibiOS""
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\PX4FirmwareUpgradeThread.cc:204 - "_findBootloader "\\\\.\\COM14""
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\PX4FirmwareUpgradeThread.cc:249 - "Found bootloader"
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:735 - ""_buildAPMFirmwareNames description(ArduPilot) vid(4617/0x1209) pid(22337/0x5741)""
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:752 - "Board id match: "KakuteF7 - 4.0.3" "ArduPilot" ("KakuteF7-BL") "https://firmware.ardupilot.org/Copter/stable/KakuteF7/arducopter.apj" FirmwareUpgradeController::CopterFirmware"
[D] at C:\projects\qgroundcontrol\src\comm\SerialLink.cc:231 - "open failed "Access is denied." QSerialPort::PermissionError "ArduPilot ChibiOS on COM14 (AutoConnect)" true"
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:735 - ""_buildAPMFirmwareNames description(ArduPilot) vid(4617/0x1209) pid(22337/0x5741)""
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\FirmwareUpgradeController.cc:752 - "Board id match: "KakuteF7 - 4.0.4" "ArduPilot" ("KakuteF7-BL") "https://firmware.ardupilot.org/Copter/latest/KakuteF7/arducopter.apj" FirmwareUpgradeController::CopterFirmware"
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\PX4FirmwareUpgradeThread.cc:279 - "PX4FirmwareUpgradeThreadWorker::_flash"
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\PX4FirmwareUpgradeThread.cc:314 - "PX4FirmwareUpgradeThreadWorker::_erase"
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\PX4FirmwareUpgradeThread.cc:320 - "Erase complete"
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\PX4FirmwareUpgradeThread.cc:285 - "Program complete"
[D] at C:\projects\qgroundcontrol\src\VehicleSetup\PX4FirmwareUpgradeThread.cc:298 - "Verify complete"

DonLakeFlyer, 2020-03-11:
@dkemxr Thanks for much for reporting and testing. Looks like we are good now.
@tridge Thanks for the help.

DonLakeFlyer, 2020-03-11:
New stable point release should be out by tomorrow some time.

closed 2020-03-11 by DonLakeFlyer

tridge, 2020-03-12:
thanks @DonLakeFlyer !