mavlink/qgroundcontrol #8436: TCP-MPEG2 video lags/hangs the application when no video source is present

Antiheavy, 2020-02-27:
Happening in Stable v4.0 and Master.

When "TCP-MPEG2 video" source is selected, an IP and port are entered, but there is no video stream present, the application freezes and hangs/lags severely.

When a TCP-MPEG2 video source is actually present, the application works fine.

Same behavior tested on Windows and a manual iOS build:
[image]

[image]

To reproduce:
1) Open QGC
2) no not have a video source present
3) change the video source in general settings to TCP-MPEG2 Video
4) enter a IP and port, press enter
5) try to navigate around the app, observe severe lag and hang behavior

To get back to non-laggy state:
1) force close the application
2) edit the Qgroundcontrol.ini file to clear the tcpURL entry:
[Video]
videoSource=TCP-MPEG2 Video Stream
tcpUrl=

FYI @andrewvoznytsa @dakejahl

andrewvoznytsa, 2020-02-28:
Worked well for me if stream source was 127.0.0.1 and host system was Ubuntu. Other cases are really laggy. Will fix.

Antiheavy, 2020-02-28:
@andrewvoznytsa for what it is worth the version of QGC you had me test with our TCP-MPEG2 hardware on Feb 21st works fine both with a video stream and without a video stream.
[image]

andrewvoznytsa, 2020-02-28:
@Antiheavy this version which you tested was using some trick to workaround 'no stream' issue. It did not work properly in some conditions (if Microhard modems were used, for example) at all. I tried to fix that issue using some queued messaging - that looked like proper solution for all cases. Now it is clear that it needs some tweak (probably extra thread) to make it stable and working well in all cases.

It also means reminds me that we need proper CI testing for video stuff...

Antiheavy, 2020-02-28:
> Microhard modems were used

.. we happen to be using microhard modems in our system, although not in the way that QGC has native support for them, the UDP command and control link and TCP MPEG2 video run through a router first....

andrewvoznytsa, 2020-02-29:
@Antiheavy please confirm that https://ci.appveyor.com/api/buildjobs/xgcpylfx0cu93rqy/artifacts/QGroundControl-installer.exe works not worse than old builds.

Antiheavy, 2020-03-01:
I’m out of town for the weekend but will test monday.

The1only, 2020-03-01:
Moving from v3.5.6 to v4.0 the video quality for UDP Stream went down the drain, and also the stability, not the system tends to hang.

Mac OSX

andrewvoznytsa, 2020-03-01:
@The1only what do you mean saying 'the video quality for UDP Stream went down the drain'? Don't want to guess - if you give me details then I can help, perhaps. But please create separate ticket if it is not related to 'TCP-MPEG2 video lags' - will be much easier to manage that.

The1only, 2020-03-01:
Ok sorry, this is a UDP H.264 lag/quality  issue. I whent back to 3.5.6 and all is OK. 
I will make a propper ticket or two.

Antiheavy, 2020-03-02:
> @Antiheavy please confirm that https://ci.appveyor.com/api/buildjobs/xgcpylfx0cu93rqy/artifacts/QGroundControl-installer.exe works not worse than old builds.

yes, the build you linked does work "not worse than the old builds" as far a I could tell with my hardware testing in the lab.  thanks!

Antiheavy, 2020-03-03:
@andrewvoznytsa am I (or you) good to close this now that https://github.com/mavlink/qgroundcontrol/pull/8448 is merged?  I also successfully tested this on today's "Stable_V4.0-11852" build from QGC appveyor and the lag is no longer there either.

andrewvoznytsa, 2020-03-03:
Fixed by https://github.com/mavlink/qgroundcontrol/pull/8448

closed 2020-03-03 by andrewvoznytsa