mavlink/qgroundcontrol #8437: Feature request: option to override Checklist position in custom plugin

Antiheavy, 2020-02-28:
We use the Checklist function in QGC with our custom plugin.  The new QGC 4.0 has reworked the checklist quite a bit (some nice new features!), including having it be centered on-screen.  

At first impression, a centered checklist is great because it is right there in your face and you can't help but notice it.  However, the centered on-screen checklist typically covers up most of the mission, the Home location, and the vehicle.  Our custom checklist asks the user to verify the vehicle location and heading on the map as well as check the takeoff and landing (fixed wing) are planned correctly and clear of obstacles.  The centered checklist is problematic for this.

@dogmaphobic how hard would it be to get some tools to be able to let plugins move the checklist to other locations onscreen?  Or at least move it back to "dock" in and out of the Checklist button they way it did before QGC 4.0?

[image]

FYI @dakejahl

DonLakeFlyer, 2020-02-28:
Yeah, I wasn't thinking about this when I saw the change. The keep the center of the screen clear concept was one of the reasons for the connected Drop Panel stuff. @dogmaphobic  can you move this back to the way it was? Is there some reason that is not possible now?

dogmaphobic, 2020-03-03:
I guess it makes sense for the checklist to be the exception because of this one detail. The idea is that these dialogs should be the _center_ of your attention. Not something to the side.

Antiheavy, 2020-03-03:
> The idea is that these dialogs should be the _center_ of your attention.

I think this is a valid idea for many (maybe even most) use cases.  Our custom checklist has users verifying things on the map which is why we are hoping for a mechanism for the plugin to override placement.

DonLakeFlyer, 2020-03-03:
@dogmaphobic Can y ou put this back to where it was before. I'm still in hand to hand combat with other stable problems.

dogmaphobic, 2020-03-03:
Yup, I will see if I can get this in later this evening.

closed 2020-03-04 by dogmaphobic