mavlink/qgroundcontrol #8439: Feature Request: Add loiter direction option to mission planning wizard

kosakwayne, 2020-02-28:
When using the mission planning wizard, after placing the landing location the "loiter clockwise" option should be present.  This is the only setting that needs to be consistently changed with a standard default landing procedure. 
I have sketched up a mockup below of what it could look like:

[image]

DonLakeFlyer, 2020-03-01:
My concern here is that this is creating a parallel ui to the normal ui. Which wasn't the original intent of this wizard mode stuff. It was just meant to handle completely missing information like no polygon. I understand why this is being asked for. But I think it possibly point to a larger problem which is the the auto-move to the next item when you click done may not be a good idea. I think it may be better to stay on the item so you can tweak as you need before finishing up the rest of the incomplete items.

M-Skelton, 2020-04-06:
I've used the planning wizard a bunch now. I think this is a key feature that is missing. I think most users won't touch any of the rest of the landing parameters when working with commercial solutions. However, flipping the loiter direction will be used.

closed 2025-04-17 by DonLakeFlyer