mavlink/qgroundcontrol #8440: Geo-Fence supports ArduCopter 4.0 but not ArduPlane 4.0

nnewsted, 2020-02-28:
Tell us a bit about the feature:
- What problem does it solve?
    Geo-Fence enabled for ArduPlane
- What does it do?
   Enabled Geo-Fence for ArduPlane
- What flight stacks must it work with? (All, PX4, ArduPilot)
   ArduPilot - Plane 4.0 or higher
- Relevant vehicle types? (All, multirotor, fixed-wing, VTOL, submarine, etc.)
   fixed-wing
- Are there other systems that have this feature?
  Yes, ArduPilot - Copter
- What communication/integration standards does the feature rely on (e.g. MAVLink commands etc.)
   
Any additional context you can provide will make the feature easier to evaluate (e.g. mockups, detailed specification, etc.)

DonLakeFlyer, 2020-02-28:
Turn on VehicleLog by doing this: https://docs.qgroundcontrol.com/en/SettingsView/console_logging.html and attach the console here. And then also save the telemetry log for initial vehicle connection and link to it here.

That will help figure out what is going on.

DonLakeFlyer, 2020-02-28:
@tridge Here is what I see testing with a PixRacer board and ArduPlane 4.0 over usb connection:
* Connect vehicle usb first, then boot QGC: Vehicle reports rally support in capabilities bits but not geofence
* Boot QGC first, then connect vehicle usb: Vehicle acks MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES with ok. But it never sends an AUTOPILOT_CAPABILITIES.

DonLakeFlyer, 2020-02-28:
> * Boot QGC first, then connect vehicle usb: Vehicle acks MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES with ok. But it never sends an AUTOPILOT_CAPABILITIES.

ArduCopter shows this same problem

nnewsted, 2020-02-28:
Don,

I've attached the log from QGC. I am running plane SITL, (JSBSim Rascal) on a remote computer. I built ArduCopter from the Plane4.0 branch, which is version V4.0.2-rc3, and I was able to plan a geofence and rally points.

[QGC_Log.txt](https://github.com/mavlink/qgroundcontrol/files/4269498/QGC_Log.txt)

DonLakeFlyer, 2020-02-28:
Ok, so ArduPlane SITL always shows this problem: Vehicle acks MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES with ok. But it never sends an AUTOPILOT_CAPABILITIES. Which then doesn't tell QGC anything about what is supported.

Need to wait to here from @tridge. From a Mavlink protocol standpoint QGC is doing all the right things.

nnewsted, 2020-03-01:
It seems that ArduPlane is the issues. I'll open an issue on the ArduPilot page.

DonLakeFlyer, 2020-03-01:
> ArduPlane SITL always shows this problem: Vehicle acks MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES with ok. But it never sends an AUTOPILOT_CAPABILITIES. Which then doesn't tell QGC anything about what is supported.

I'm running SITL like this:
```
build/sitl/bin/arducopter --model quad --defaults Tools/autotest/default_params/copter.parm --instance 1 --home 47.6333022928789,-122.08833157994995,19,0
```

DonLakeFlyer, 2020-03-01:
Re-opening just the track the capabilities problem

DonLakeFlyer, 2020-03-05:
QGC wil now retry for AUTOPILOT_VERSION multiple times. It also switch to mavlink v2 in the ArduPilot case where PROTOCOL_VERSION is not supported but capability bit for v2 is set.

closed 2020-03-05 by DonLakeFlyer