mavlink/qgroundcontrol #8454: Analize plugin missing feature

diablo1281, 2020-03-02:
Hi,
I have just upgrade to version 4.0.0 of QGCS and notice that MAVLink Inspector and Analize plugin get merged. There is a problem however. You can plot only 4 items at once and can't plot specific item from table variable (like field `voltages' from message `BATTERY_STATUS`). I am developing autopilot software and these features are very helpful.

Any chance to bring at least analize tool back?

dogmaphobic, 2020-03-03:
Indeed, the current graphing does not handle arrays of values. It would be quite tricky to add that. With that said, the theoretical limit is two separate plots with 6 values each.

DonLakeFlyer, 2020-03-03:
> not handle arrays of values

Did that work before. Don't remember.

dogmaphobic, 2020-03-03:
Me either but I guess it did, otherwise this wouldn't be brought up. The vast majority of the cases, each field contains one value. Things like battery (cells), quaternions, etc. have an array of values for a single field. The code doesn't handle that at all.

DonLakeFlyer, 2020-03-03:
Isn't it just another level to the hierarchy?

diablo1281, 2024-08-23:
Hi, any news about bringing this feature back?

diablo1281, 2025-05-15:
We are approaching V5 and the same problem persists...