mavlink/qgroundcontrol #8462: Center on "Mission" and Center on "All items" zooms to world

Antiheavy, 2020-03-03:
Testing a "Stable" build from appyveyor (Stable_V4.0-11852) I noticed that Plan view "Center map on:" "Mission" and "All items" cause the map to zoom out to the world and center on somewhere that looks suspiciously like Switzerland ;-)

[image]

dogmaphobic, 2020-03-03:
LOL

DonLakeFlyer, 2020-03-03:
I found another bug in the mapFitFunctions today. Which I fixed this morning. My guess is this is good now. I'll take another look tomorrow to see if it works now.

dogmaphobic, 2020-03-03:
As far as I know, only Mocklink has Zürich hardcoded. If this is indeed the case.

DonLakeFlyer, 2020-03-03:
> Mocklink has Zürich hardcoded

yup. But you can make it be my house too if you change the ifdef :)

kosakwayne, 2020-03-03:
This also happens when loading a KML/SHP for a survey

DonLakeFlyer, 2020-03-03:
Can you test with latest stable: https://qgroundcontrol.s3-us-west-2.amazonaws.com/builds/Stable_V4.0/QGroundControl-installer.exe?

DonLakeFlyer, 2020-03-03:
Daily builds aren't always fully up to date with the in progress stable point release.

kosakwayne, 2020-03-03:
@DonLakeFlyer I'm seeing the same behavior when loading KML/SHP for survey in the latest stable you just linked.

Antiheavy, 2020-03-03:
> I'm seeing the same behavior when loading KML/SHP for survey in the latest stable you just linked.

yep!  I'm seeing this too on Stable from 21 hours ago:
[image]

I'll try the one @DonLakeFlyer linked above next.

Antiheavy, 2020-03-03:
> Can you test with latest stable: https://qgroundcontrol.s3-us-west-2.amazonaws.com/builds/Stable_V4.0/QGroundControl-installer.exe?

still happening with this stable release too.

DonLakeFlyer, 2020-03-03:
Can you try again. New build is there.

Antiheavy, 2020-03-03:
> Can you try again. New build is there.

Fixed in the newest Appveyor build:  Stable_V4.0-11859

Thanks!

closed 2020-03-03 by Antiheavy