mavlink/qgroundcontrol #8466: QGC v4.0.0 one Windows 10 - Downloads and installs but fails to execute

mcsauder, 2020-03-03:
Hi @DonLakeFlyer,

I am able to download and install the latest QGC v4.0.0 but the executable fails to run properly once installation is complete.  Below is a series of screenshots demonstrating the steps to reproduce and the result.  I have observed this same behavior on 2 Dell latitude ATG laptops with new windows 10 installations and all OS updates applied.  I have also observed this behavior on an Asus ZenBook pro with similar results.

I am able to install and run QGC v3.5.6 on each of the machines that v4.0.0 fails with.

Thanks for any advice you have!

[image]
[image]
[image]
[image]
[image]
[image]
[image]

(From the same machine as the screenshots above after uninstalling and reinstalling v3.5.6)
[image]

DonLakeFlyer, 2020-03-03:
Follow the instructions here: https://docs.qgroundcontrol.com/en/Support/Support.html for finding the crash dump file. Zip that up, put it somewhere and link to it here.

DonLakeFlyer, 2020-03-03:
Hmm, that might now be a crash. So you may not have any dumps generated.

DonLakeFlyer, 2020-03-03:
It doesn't look like it but are you running 32 bit version of WIndows?

DonLakeFlyer, 2020-03-03:
Don't worry about the crash dump. I figured out what is going wrong from the error code.

DonLakeFlyer, 2020-03-03:
@mcsauder Can you install a new build from here: https://qgroundcontrol.s3-us-west-2.amazonaws.com/builds/Stable_V4.0/QGroundControl-installer.exe and let me know the results.

mcsauder, 2020-03-03:
Hi @DonLakeFlyer, That one works!  It makes sense that the issue was hitting new computers that were missing parts of the .NET framework that other program installations could previously have installed on some machines.

Thanks so much for your help!

closed 2020-03-03 by mcsauder

DonLakeFlyer, 2020-03-04:
@mcsauder Thanks for reporting and testing