mavlink/qgroundcontrol #8474: TCP video doesn't work on qgroundcontrol

MarwanFCT, 2020-03-04:
Hello everyone, i’m sending telemetry data from my navio2+raspberrypi3 to my android device (qgroundcontrol) on tcp protocol, when i sent the video over 4g on udp protocol it greyed out and pixelated a lot (for what i’ve read due 4g connection not being fast enough) so i tried out TCP protocol for video, but this time nothing shows up. I don’t know what’s wrong, sending tcp video to mission planner works, but not for qgroundcontrol. I guess this might be a bug of the app itself. That's why im posting here.
Thanks for any hints on what the problem is!

andrewvoznytsa, 2020-03-04:
It is not clear which QGC do you use - can you please try 3.5 stable and 4.0 stable and let me know if one of them works for you?

MarwanFCT, 2020-03-05:
I've tried 3.6 4.0.1 4.0 but none of them worked for tcp video streaming.

andrewvoznytsa, 2020-03-05:
So it isn't because of one of latest changes (I had some ideas).

Now I need more information.

You can help me if you tell me how can I simulate your video setup. For example - if you are running gstreamer on vehicle side then show me please command line. If you do something else - please provide as much information as you can. I'm interesting to see H.264 encoder options, mux/packetizer settings and anything else related to transport. Also I'd see what URL do you pass to QGC.
Perhaps it will be useful if you send us your video receiver and gstreamer logs.

We need to start somewhere - and so better your bug description will be so faster I'll fix that. As for now I can only guess that you are running QGC on Windows, probably Win 10. Less guessing will help. Thanks.

MarwanFCT, 2020-03-10:
Yeah sure, i don't really know how to view the gstreamer logs but i can send the pipeline im using for a RaspiCam: raspivid -n -t 0 -rot 180 -w 640 -h 480 -fps 30 -b 600000 -co 60 -sh 40 -sa 10 -o - | gst-launch-1.0 -e -vvvv fdsrc ! h264parse ! rtph264pay pt=96 config-interval=5 ! tcpserversink host=192.168.192.65 port=5600. You are right i specified very little, my apologies. I'm in fact using Windows 10 as the receiving end with qgroundcontrol, the url im introducing in qgroundcontrol for TCP is: 192.168.192.65:5600 (i followed the example showed when clicked on the question mark help button)

andrewvoznytsa, 2020-03-10:
There is problem with pipeline - in tcp case we assume to use
mpegtsmux/demux (MPEG-2 transport stream) instead of RTP packetizers.

Personally I don’t like both solutions and I’d prefer if people would use
RTSP with RTP over TCP as more standard way to send media.

Can I ask you why do you prefer TCP over UDP?

MarwanFCT, 2020-03-11:
How would the pipeline end like with the MPEG-2 transport stream? if you don't mind me asking. I prefer TCP over UDP because as i said on my first post the UDP one greys out (happened to more people with QGroundControl) and it happened to me with mission planner too, it seems like when using 4G network instead of wifi that happens with UDP. So i tried TCP with mission planner (with that same pipeline) and it worked just fine, so i figured TCP would work fine for QgroundControl but encountered this other problem. 
Thank you!

andrewvoznytsa, 2020-03-11:
`gst-launch-1.0 videotestsrc pattern=ball ! video/x-raw,width=640,height=480 ! x264enc ! mpegtsmux ! tcpserversink host=192.168.192.65 port=5600` will work for you.

And as for someone who is testing TCP in the field - do you feel that delay is increasing? Is not it making problems for you? Or do you use some kind of slow vehicle and delay is not so critical for you?

I'm trying to learn real use cases and improve video - your information could be very helpful.

@DonLakeFlyer I think this issue can be closed - it was QGC misuse.

MarwanFCT, 2020-03-11:
I did get +1second delay at first but with the same pipeline i used here for mission planner i get 300-400ms delay. Delay is kind of really important to me, i would use UDP for latency improvement if it worked correctly on QgroundControl or mission planner. Since i had an erroneous pipeline i will try this one and get back to you about how latency goes with it. 
Thanks!

MarwanFCT, 2020-03-11:
> `gst-launch-1.0 videotestsrc pattern=ball ! video/x-raw,width=640,height=480 ! x264enc ! mpegtsmux ! tcpserversink host=192.168.192.65 port=5600` will work for you.
>
> And as for someone who is [...]

Thanks for the pipeline, but since i'm using raspicam as source i have to use fdsrc, and that doesn't link with the rest of the pipeline or am i not understanding this well?

andrewvoznytsa, 2020-03-11:
This pipeline is just test case to verify that TCP streaming works (something known to work). You can use it to build your RPi pipeline - probably something like `raspivid -n -t 0 -rot 180 -w 640 -h 480 -fps 30 -b 600000 -co 60 -sh 40 -sa 10 -o - | gst-launch-1.0 -e -vvvv fdsrc ! h264parse ! mpegtsmux ! tcpserversink host=192.168.192.65 port=5600`

MarwanFCT, 2020-03-11:
So on Qgroundcontrol im just supposed to select TCP-MPEG2 and the url 192.168.192.65:5600 right?

andrewvoznytsa, 2020-03-11:
Right.

MarwanFCT, 2020-03-11:
[image]
Still not getting the stream, with this same command on version 4.0 of qgroundcontrol

andrewvoznytsa, 2020-03-11:
You need to try with my test pipeline first - run it on some PC and adjust IP address.

andrewvoznytsa, 2020-03-11:
And make sure that you use latest 4.0.

MarwanFCT, 2020-03-11:
I'm not sure what you mean by running the test pipeline on a pc ? on my windows 10?

andrewvoznytsa, 2020-03-11:
Yes, why not? Just install gstreamer there and run that pipeline (adjusting IP of course) to make sure that TCP streaming works for you. If yes - we can move to your RPi. It is easiest way to find where problem is - on sender or receiver side and then try to fix it. A few guys are using TCP and they report that it is working (latest 4.0.x).

MarwanFCT, 2020-03-11:
Okay, will do tomorrow, thanks for your help, support and patience, i'm not experienced with gstreamer^^. When i try it i will report back to you ASAP.

DonLakeFlyer, 2020-03-11:
Reopen if this proves out to be a QGC problem instead of a setup problem.

closed 2020-03-11 by DonLakeFlyer