mavlink/qgroundcontrol #8478: Waypoint numbers optional

sanderux, 2020-03-04:
Although the view is much cleaner, external validators and phone support on mission plans makes the lack of waypoint numbers very impractical.

I propose an on/off switch, setting or showing the numbers in the waypoint list.

Saijin-Naib, 2020-03-05:
Agree with this. I like being able to validate the order without having to open the plan in a text editor.

DonLakeFlyer, 2020-03-05:
> validate the order

@Saijin-Naib Can you explain some more what you mean by that. Why do you need to validate the order?

Saijin-Naib, 2020-03-05:
OCD-ness, not trusting the generated plans, ensuring that the transects flow the way I expect them to (especially important on cross-grid), etc. etc. etc.

I'd really love a means to inspect each individual waypoint/command generated for a Survey as well to validate it's properties...

DonLakeFlyer, 2020-03-05:
Got it. Thanks for the info.

DonLakeFlyer, 2020-03-05:
> I'd really love a means to inspect each individual waypoint/command generated for a Survey as well to validate it's properties...

FYI: You can get that if you download the Plan back from the vehicle after you upload. Then it will degrade to individual commands.

Saijin-Naib, 2020-03-05:
> FYI: You can get that if you download the Plan back from the vehicle after you upload. Then it will degrade to individual commands.

Thanks for the tip/work-around. That's very valuable. An in-app way to do the same would likely be easier, but this suits for now.

DonLakeFlyer, 2020-03-05:
Also for folks trying to deal with no waypoints numbers. You can see waypoints from the Terrain display (Click T at bottom of Plan). Those item are also clickable to set current item.

Antiheavy, 2020-03-14:
We had some of our users start trialing the newest QGC Stable this week and the first peice of feedback was "where did the waypoint numbers go? can I enable them somehow? we use those to communicate to eachother about where the aircraft is."

I was planning on seeing if people would even notice the numbers where gone... turns out they noticed right away.  ha!

I think a Fly View Settings option would be pretty useful for this.

DonLakeFlyer, 2020-03-14:
> we use those to communicate to each other about where the aircraft is.

That's a major problem which I'm getting feedback on from multiple places. 

At this point I even wonder whether a show.hide thing is feasible with default to off. If majority of people end up turning that on, then it's a failure.

sanderux, 2020-03-21:
@DonLakeFlyer 
I believe that, in professional practice, waypoint numbers are critical. Perhaps you could reconsider the decision, or even call the test run a failure and bring them back.

At a minimum, I would hold off our users from the current QGC version until there is an option to switch them back on. The workflow is simply too complicated without them now.

DonLakeFlyer, 2020-03-21:
Sorry didn't close this. I put them back a few days ago: https://github.com/mavlink/qgroundcontrol/pull/8568

closed 2020-03-21 by DonLakeFlyer