mavlink/qgroundcontrol #8484: 4.0 android/mobile qml issues

jaxxzer, 2020-03-05:
In each of these screenshots, some item or text is too wide and is being cropped

[image]

[image]

[image]

[image]

tcanabrava, 2020-03-05:
some of those are also happening on desktop since 3.x

DonLakeFlyer, 2020-03-05:
What size device. Target mobile is 10" tablet. Although things will sort of work below that we generally don't try to fix layout issues on smaller devices.

jaxxzer, 2020-03-05:
> What size device. Target mobile is 10" tablet. Although things will sort of work below that we generally don't try to fix layout issues on smaller devices.

Oh, it's an 8" device. I did not know about the 10" thing.

DonLakeFlyer, 2020-03-05:
There is some small amount of write up on it here: https://dev.qgroundcontrol.com/en/ui_design/multi_device_pattern.html

jaxxzer, 2020-03-09:
noting here for @tcanabrava to check the font size options in the general settings menu, that can throw some of the layouts off too.

jaxxzer, 2020-05-13:
@Williangalvani merged pull request has fixed the problem on linux. The problem persists on windows.

closed 2024-06-02 by HTRamsey