mavlink/qgroundcontrol #8485: Missing game controller button mappings

jaxxzer, 2020-03-05:
Controller buttons are 'normalized' with hard-coded mappings contained in [gamecontrollerdb.txt](https://github.com/mavlink/qgroundcontrol/blob/master/resources/SDL_GameControllerDB/gamecontrollerdb.txt). We are missing some mappings from our 'supported' controllers.

The official SDL mappings are [here](http://hg.libsdl.org/SDL/file/88068887c25d/src/joystick/SDL_gamecontrollerdb.h).

Not correctly mapped in QGC:

Linux:
- xbox one connected by usb
- ps4 connected by usb

Android:
- ps4 connected by bluetooth

jaxxzer, 2020-12-02:
Here is the result on Linux with PS4 controller. There are two problems:
1. a calibration should not be required (a default calibration should be included)
2. only 16 buttons should be available (there are 17)

[image]

DonLakeFlyer, 2020-12-02:
Is there hard-code mapping capability in QGC? If so, didn't realize that was there.

jaxxzer, 2020-12-09:
> Is there hard-code mapping capability in QGC? If so, didn't realize that was there.

Yes, the relevant file in this repository is [gamecontrollerdb.txt](https://github.com/mavlink/qgroundcontrol/blob/master/resources/SDL_GameControllerDB/gamecontrollerdb.txt)

clydemcqueen, 2023-04-01:
Possibly related? I have 2 xbox one controllers on Linux, connected by USB. They both test fine in other apps. In QGC the left trigger button, which I expect to be button 9 "mount_tilt_down" in ArduSub, does not work.

From `journalctrl -f:`
~~~
New USB device found, idVendor=045e, idProduct=02dd, bcdDevice= 2.03
~~~

Is the fix a matter of copying [this line](https://github.com/libsdl-org/SDL/blob/07000bfc331448ce03e56b67de45cde3eb9e45aa/src/joystick/SDL_gamecontrollerdb.h#LL388C21-L388C25) to the appropriate spot in QGC source and re-building?

HTRamsey, 2024-06-02:
A newer sdl and newer mappings have been added that should resolve this

closed 2024-06-02 by HTRamsey